ArduPilot/ardupilot_wiki #131: Create mechanism to control whether common pages are copied to a particular wiki

hamishwillee, 2015-08-26:
Some common pages are not relevant to specific wikis - for example UAS Testing Centres and Terrain following are not relevant to Rover. The aim of this post if to 

* determine a mechanism for marking some common articles as "do not copy" so that they aren't copied.
* Identify [common articles](http://rover.ardupilot.com/wp-admin/edit.php?orderby=title&order=asc&s=common-&post_status=all&post_type=wiki&action=-1&m=0&paged=1&action2=-1) that can't be copied to Rover

hamishwillee, 2015-08-26:
So proposal is to add shortcode with format like this:
```
[copywiki source="planner" destination="rover,plane"]
```

The **copy_common.py** script would always copy the file if there was no shortcode. If there is shortcode AND the destination exists, then it would check the string before copying. So to stop copying rover documents you would do
```
[copywiki source="planner" destination="copter,plane"]
```
The script won't delete a common page already on the target wiki. In version 1 it will ignore the source (that is kind of future proofing)

@gmorph I'm trying to work out a good method for preventing unwanted copying of common docs (primarily to rover). Does this sound sane for users - do nothing for existing behaviour, but if you want to stop publishing to rover you can add the above shortcode without rover in it.

gmorph, 2015-08-26:
Yes I think that's appropriate.  Very good idea Hamish!

Thanks, Grant.

hamishwillee, 2015-08-27:
This is in place now - the wordpress scripts will not copy to a destination if it has a copywiki and the target is not in the "destination" field.

gmorph, 2015-08-27:
That's great.  Nice work Hamish.

Thanks, Grant.

hamishwillee, 2015-08-28:
@gmorph Cool. I've sent you an email about the copy script, but you really can ignore it if you choose.

What I do need from you is a list of articles to exclude from the rover wiki - I which of these are really in your opinion not common and confusing for Rover users: http://rover.ardupilot.com/wp-admin/edit.php?orderby=title&order=asc&s=common-&post_status=all&post_type=wiki&action=-1&m=0&paged=1&action2=-1

closed 2015-08-31 by hamishwillee