ArduPilot/ardupilot_wiki #136: Identify common articles that shouldn't got to rover

hamishwillee, 2015-08-31:
This is the [list of common articles](http://rover.ardupilot.com/wp-admin/edit.php?orderby=title&order=asc&s=common-&post_status=all&post_type=wiki&action=-1&m=0&paged=1&action2=-1)

This follows on from #131

davidbuzz, 2016-11-15:
@gmorph  - any idea what this issue is about?

gmorph, 2016-11-15:
Yes it was to remove link's in the Rover wiki to pages that aren't relevant for Rover.  I think most of these have been addressed.  We can close this and just address ones we find on a case by case basis.

closed 2016-11-15 by gmorph

hamishwillee, 2016-11-15:
Yes, that's right. When I did this I hoped for a quick scan through of the docs to confirm that the displayed Rover set was OK. I don't believe that happened, but it's Grants call whether to go with "methodical" or "spot fix".

gmorph, 2016-11-18:
I do go through the Rover documents quite regularly and haven't had anything jump out at me as being a problem so I think closing this is ok.  
Thanks, Grant.

hamishwillee, 2016-11-18:
Sounds good to me! Thanks