ArduPilot/ardupilot_wiki #138: TKOFF_TDRAG_SPD1: Conflicting documentation

Georacer, 2015-09-01:
Parameter documentation reads: "It should also be set to zero for tricycle undercarriages."

However, one parameter above, in `TKOFF_TDRAG_ELEV`we read:
"For most tricycle undercarriage aircraft a value of zero will work well, but for some tricycle aircraft a small negative value (say around -20 to -30) will apply down elevator which will hold the nose wheel firmly on the ground during initial acceleration."
meaning that a non-zero `TKOFF_TDRAG_SPD1` value is needed for `TKOFF_TDRAG_ELEV` to take effect.

Isn't that a conflicting statement?

hamishwillee, 2015-09-01:
Hi @Georacer 

Thanks very much for posting.Yes, this statement assumes that TKOFF_TDRAG_SPD1 will be zero for a tricycle undercarriage. In most cases it will be.

However we know from the TKOFF_TDRAG_ELEV can be less than zero, and if we set TKOFF_TDRAG_SPD1  to zero in this case, we will simply skip through the tail/nosewheel drag step. 

Perhaps a change in [TKOFF_TDRAG_SPD1](http://plane.ardupilot.com/wiki/arduplane-parameters/#takeoff_tail_dragger_speed1_arduplanetkoff_tdrag_spd1) from "It should also be set to zero for tricycle undercarriages." to "This should normally be set to zero for tricycle undercarriages unless TKOFF_TDRAG_ELEV<0 (see TKOFF_TDRAG_ELEV for additional information)."

@gmorph What do you think?

gmorph, 2015-09-08:
Fixed in this PR.  Closing.
https://github.com/diydrones/ardupilot/pull/2817
Thanks!

closed 2015-09-08 by gmorph