ArduPilot/ardupilot_wiki #141: CAMremote

ledelec, 2015-09-03:
Hello folks,

Is there any possibility to do cooperation with you and offer camera triggering device called CAMremote on your web page?

The CAMremote supports almost every camera which have remote control option available and offers control over camera main functions like AF, shutter release, movie recording and also offers possibility to control camera parameters.

We are working together with several DIY drones users and implemented suitable support to CAMremote in order to work with Pixhawk/Ardupilot systems.

http://vp-systems.eu/camremote.html

Please let me know if you are interested to do cooperation, thanks!

Linnar Edesi
VP-Systems

gmorph, 2015-09-03:
We can include it on the webpage once it has been confirmed as working.  We get a number of requests from people with devices saying hey put this on your page please.  We are happy to do so but only after we confirm something is working and provides the advertised functionality.
So, you either need to send one to one of the developers who will commit to working with you and get it up and running OR provide evidence of enough community usage of the device that we know its a compatible and functioning device.

Let us know.

Thanks, Grant.

hamishwillee, 2015-09-08:
It sounds like this CAM integration is still "work in progress". I am closing this until there is a document task that can be performed.

closed 2015-09-08 by hamishwillee

rmackay9, 2015-09-08:
@ledelec, I think we just need to get a couple of these devices into the hands of some people on the dev team who have the time to verify that they work.  I'm not sure who those people are but I'd start with @CraigElder who is well placed to figure out who those people could be.

ledelec, 2015-09-08:
> comment-138434823

Hello Randy,

Thanks for reply!

I'm happy to ship CR to mentioned people if I could get address where to
ship and which cameras they would like to test.

The CAMremote has been on marked almost 6 years and should be quite well
known. We have been selling these several thousands already. I could
provide some demo videos to proof its behavior with several cameras if
needed.

We are launching Wifi bridge for CAMremote in few weeks which allows to
control even more cameras with Wifi support. Also CAN-bus support is
coming. I believe CAMremote is remote control platform which is supporting
almost any camera which have remote control feature supported.

Best Regards,
Linnar

CraigElder, 2015-09-08:
I have been in touch with Linnar and I have asked Stefan (both are in Finland so that should save on the shipping costs) if he would be interested in testing the board and report back to us.