ArduPilot/ardupilot_wiki #142: No source code for px4flow-klt-06Dec2014.px4

jlian, 2015-09-04:
[common-PX4FLOW Optical Flow Camera Board Overview](http://copter.ardupilot.com/wiki/common-px4flow-overview/) - This article links to [PX4Flow sensor firmware using Lucas-Kanade method for use with ArduPilot](http://ardupilot.com/downloads/?did=118) which is apparently licensed under GPL. I'm in need for the source code of this file, but cannot find it anywhere.

gmorph, 2015-09-05:
Is this what your looking for?
     https://github.com/PX4/Flow
Thanks, Grant

closed 2015-09-05 by rmackay9

hamishwillee, 2015-09-05:
@rmackay9 Before we close this (any) post we should consider if there is a docs requirement - do we need to add this link to the source from this doc?

jlian, 2015-09-05:
That is the official firmware by PX4 but not actually the code I'm looking for. From the linked documentation:

> We recommend you upgrade to this modified PX4FLOW sensor firmware. using QGroundControl. This modified firmware uses a different method to determine the optical flow and provides readings over a wider range of light and distance, but has a smaller maximum flow speed compared to the standard firmware.

It's not the same as the official firmware, and is only available in binary.

gmorph, 2015-09-05:
Ok then I don't know where that firmware is.  I suggest you join the PX4 gitter channel and ask in there.
Thanks, Grant.

rmackay9, 2015-09-06:
I'm pretty sure the code is from some point in the px4 firmware tree.  Paul provided me with the binary so I'll ask him if he's got the repo's hash tag that he used.

jlian, 2015-09-06:
Thanks, Randy!

priseborough, 2015-09-09:
Binary was compiled from my branch here:

https://github.com/priseborough/px4flow/tree/klt_flow

jlian, 2015-09-09:
Thanks @priseborough! @rmackay9 do you think it would be good to add the github link to the docs?

hamishwillee, 2015-09-10:
@jlian Yes. @rmackay9 made the addition last night (thanks Randy!). This is now (even more) closed!