ArduPilot/ardupilot_wiki #146: Typo in ArduPlane parameters

Georacer, 2015-09-14:
Hello everyone,

I know that parameter documentation is embedded in the code, but I think this is the appropriate place to raise this issue. If not, let me know.

Short Failsafe Action reads:

>Short failsafe action (ArduPlane:FS_SHORT_ACTN)
>The action to take on a short (FS_SHORT_TIMEOUT) failsafe event. A short failsafe even can be triggered either by loss of RC control (see THR_FS_VALUE) or by loss of GCS control (see FS_GCS_ENABL). A short failsafe event in stabilization and manual modes will cause an change to CIRCLE mode if FS_SHORT_ACTN is 0 or 1, and a change to FBWA mode if FS_SHORT_ACTN is 2. In all other modes (including AUTO and GUIDED mode) a short failsafe event will cause no mode change is FS_SHORT_ACTN is set to 0, will cause a change to CIRCLE mode if set to 1 and will change to FBWA mode if set to 2. Please see the documentation for FS_LONG_ACTN for the behaviour after FS_LONG_TIMEOUT seconds of failsafe.
>0 	Continue
>1 	Circle/ReturnToLaunch
>2 	Glide

If I read the documentation correctly, it says that when in STAB or MANUAL:
0 = CIRCLE
1 = CIRCLE
2 = FBWA
and when in any other mode (including AUTO)
0 = NO CHANGE
1 = CIRCLE
2 = FBWA

However, as you can see, the corresponding table in the documentation is completely contradictory. It contains an RTL where none is there to be found and a "Glide" mode which is unknown and undocumented.

I think some corrections are due here.

Thanks!

gmorph, 2015-09-14:
When Tridge wrote Glide he really means what will happen when it goes into FBWA mode - the plane will simply glide.  But your right - we don't really mention glide anywhere else so I'll change it over.

Georacer, 2015-09-14:
Nice!

gmorph, 2015-09-14:
Resolved with this PR
https://github.com/diydrones/ardupilot/pull/2848
Note I read the code and confirmed this is the behaviour.  Its in ArduPlane/events.cpp failsafe_short_on_event(enum failsafe_state fstype) if anyone else is interested.
Georacer - keep spotting those doco problems!!  Thanks!!
Thanks, Grant.

closed 2015-09-14 by gmorph

hamishwillee, 2015-09-14:
@Georacer Thanks. Yes, this is the right place for any wiki issue (content or infrastructure). In this particular case you could also post a [ardupilot github issue](https://github.com/diydrones/ardupilot/issues) because, as you say, parameters and their documentation are part of the code.

Thanks for fixing Grant!