ArduPilot/ardupilot_wiki #147: Switched TELEM ports in Overview

Georacer, 2015-09-14:
In this overview page: http://plane.ardupilot.com/wiki/common-pixhawk-overview/ the legend says that TELEM1 is for OD and TELEM2 is for radio telemetry.

However, I think this is the other way around, as is also seen in this quick connection page: http://plane.ardupilot.com/wiki/common-pixhawk-wiring-and-quick-start/#pixhawk_wiring_chart and http://plane.ardupilot.com/wiki/common-minim-osd-quick-installation-guide/

Can someone verify? Thanks.

gmorph, 2015-09-14:
Yep you are correct George.  Wow you have great attention to detail - we've only had that picture forever and its been wrong all this time.
I'm not sure how we are going to fix that.  Hamish - are you able to edit that picture?
Thanks, Grant.

Georacer, 2015-09-14:
I'm currently going over all the parameters, since I'm trying to install a Pixhawk in an airplane with very "special nature and needs", so I want to make sure I don't miss something and mess up.

So, as a side-bonus, I point the errors. There's nothing I value more in a product other than good documentation.

hamishwillee, 2015-09-15:
Thanks very much. I fixed up the image http://planner.ardupilot.com/wiki/common-pixhawk-overview/#pixhawk_connector_assignments

closed 2015-09-15 by hamishwillee

Georacer, 2015-09-15:
Nice! It doesn't even look edited!