ArduPilot/ardupilot_wiki #148: Incosistencies in the ArduPlane arming documentation

Georacer, 2015-09-14:
Naturally, there is a wiki page for Arming setup instructions: http://plane.ardupilot.com/wiki/arming-throttle/

As we all know, this involves a few parameters, some of which are the `REQUIRE`, `CHECK` etc.
However, none of these parameters are shown in the Parameter list: http://plane.ardupilot.com/wiki/arduplane-parameters/#eeprom_format_version_number_arduplaneformat_version, expect for the `ARMING_RUDDER` parameter which is not documented in the previous page.

Has the code been updated?

Thanks.

gmorph, 2015-09-14:
Yes I have changed the code recently and I must of broken the doco.  Sorry.  I'll fix it now.  Great catch - thanks!

gmorph, 2015-09-15:
Fixed by this PR
https://github.com/diydrones/ardupilot/pull/2852
Thanks - keep them coming!

closed 2015-09-15 by gmorph