ArduPilot/ardupilot_wiki #149: Antenna Tracker needs a set-up video

hamishwillee, 2015-09-15:
Antenna Tracker needs a set-up video because a few people have been asking for this.  http://copter.ardupilot.com/wiki/common-antennatracker-introduction/

hamishwillee, 2015-09-15:
@rmackay9 This is really something you should do right?

You already created two videos but I don't think they "fit the bill". Propose adding these to a new topic in Antenna Tracker "Video Guides and Demos"?

* https://www.youtube.com/watch?v=HswYSOW-hn4
* https://www.youtube.com/watch?v=M4LxtYa94nk

davidbuzz, 2016-11-15:
I'd suggest this is a good video:   https://www.youtube.com/watch?v=K48xzI9-igA

gmorph, 2016-11-15:
Hamish are you after a build video or a how to use video?  If how to use I agree with Buzz that Randy's is fine.  However we don't have a build video that I'm aware of.
Thanks, Grant.

hamishwillee, 2016-11-15:
@gmorph A setup/build video was what was requested. Basically a bit like the series you started on Rover (and I don't know if finished). 

Either way, I still like the proposal in https://github.com/ArduPilot/ardupilot_wiki/issues/149#issuecomment-140291689 - we should collect all these videos together in a dedicated page. 

Thoughts?

brunoolivieri, 2019-07-15:
The main page have a newer video. It is not a complete assembly tutorial but it is a quite complete in terms of explanation how it works. (http://ardupilot.org/antennatracker/index.html)

Do you think that is enough and we could close the issue or it is desirable something even more complete?