ArduPilot/ardupilot_wiki #150: Simplify topic Planning a mission with Waypoints and Events

hamishwillee, 2015-09-15:
From #10 

> Replace the [Planning a mission with Waypoints and Events](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/) with a very simple page to show how to create and execute a very small mission (a take-off, one waypoint, an RTL).

hamishwillee, 2015-09-15:
@rmackay9 @gmorph I can certainly simplify this a bit- for a start I'd remove the [tips section](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/#tips). Is it not useful to include information about the auto grid or do you see that as Missionplanner specific docs?

What do you want done with all the existing videos? Just moved to section at end?

gmorph, 2015-09-16:
I have no strong feelings in this regard.  I think the page is simple
enough but if others want it simpler that's fine.

Thanks, Grant.

Hwurzburg, 2020-01-15:
seems simple enough to me...will close this issue unless someone objects strongly in a few days