ArduPilot/ardupilot_wiki #154: Second battery monitor documentation

Georacer, 2015-09-15:
Hello,

I got into reading about interfacing a 2nd battery sensor with the Pixhawk.

This seems to be implemented, since the parameters are there: http://plane.ardupilot.com/wiki/arduplane-parameters/#battery_monitoring_batt2_monitor
plus tridge had mentioned it at some beta release.

However, I'm stuck at selecting the voltage and current pins parameters.

The parameter documentation doesn't say anything about it. It only references the power port pin, duplicate from the BATT_MONITOR description.

There is no pin number list in the Pixhawk overview: http://copter.ardupilot.com/wiki/common-pixhawk-overview/

However, TCIII hinted that the pin numbers should be 13 and 14 for the two sensing pins of the 3.3V ADC here: http://diydrones.com/group/ardurover-user-group/forum/topics/how-to-install-dual-sonars-on-a-pixhawk

This took me more than 10 minutes to find and a lot of Googling. Shouldn't this information be more readily available?

Thanks.

gmorph, 2015-09-15:
The information is at the bottom of this web page
http://plane.ardupilot.com/wiki/common-pixhawk-overview/#pixhawk_analog_input_pins_virtual_pin_firmware_mapped_pin_id
which is the one referenced from the original page you noted above.
We could make it clearer by giving an example I suppose - for instance if you wanted to connect a second battery via a second power brick we could show what pins to connect where.
Thoughts?

hamishwillee, 2015-09-16:
1. @gmorph I think the link you mean is http://plane.ardupilot.com/wiki/common-pixhawk-overview/#pixhawk_analog_input_pins (other link didn't seem to go anywhere).
2. Perhaps also add update (?):

   > Virtual Pin 13 and ADC 3.3V connector Pin 4: This takes a max of 3.3V. May be used for **a second battery monitor**, sonar or other analog sensor.

   > Virtual Pin 14 and ADC 3.3V connector Pin 2: This takes a max of 3.3V. May be used for **a second battery monitor** a second sonar or other analog sensor.

3. We don't actually talk about the original BATT_MONITOR and friends or pin numbers in the documents that cover battery monitors:

   * http://plane.ardupilot.com/wiki/common-3dr-power-module/#connecting_the_pm_to_a_flight_controller_board
   * http://plane.ardupilot.com/wiki/common-using-a-current-sensor/#connection_to_pixhawk

 Instead we just show the connections and then talk about the Battery Monitor screen - which hides these details. It might be worth at least having a note that states this so that anyone doing a search can work out what is going on.

> We could make it clearer by giving an example I suppose - for instance if you wanted to connect a second battery via a second power brick we could show what pins to connect where.

I think this is a good idea because for the second battery you can't rely on the battery monitor screen in Mission planner - it doesn't cover it. What would be best is adding a section "Adding a second battery" - perhaps to these ...

   * http://plane.ardupilot.com/wiki/common-3dr-power-module
   * http://plane.ardupilot.com/wiki/common-using-a-current-sensor/#connection_to_pixhawk

Thoughts?

rmackay9, 2015-09-16:
We can also ask MichaelO to extend the MP battery monitor screen to include support for a second monitor.

hamishwillee, 2015-09-16:
Good idea Randy. Still good to document as I've suggested though.

Georacer, 2015-09-16:
@gmorph This was an oversight from my part. The analog pin list was the information I was looking for, but for some reason I wouldn't look that way.

@hamishwillee Maybe I couldn't see the info I was looking for because I was searching about battery monitor and module. Indeed, instructions for a second battery monitor would have helped me find that analog pin list.

Actually, I found myself in need of a second battery monitor in my last build, a large electric glider/payload carrier. I am a big proponent of having separate batteries for the motor (a 10S brick) and the electronics (a simple 3S). Since I deemed it more crucial to have better monitoring on the electronics battery than the motor battery, for the maiden flights, I ended up monitoring only this one.

hamishwillee, 2015-09-17:
@rmackay9 I've asked MichaelO about a mission planner update for battery 2. He is investigating, but says it looks "challenging". 

@rmackay9 @gmorph - I don't want to create the information about adding a second battery because I've not done it. Who would be the best person to update http://plane.ardupilot.com/wiki/common-3dr-power-module with information. I'd suggest a section that explains the pins you use and shows a wiring diagram with both batteries connected. That would both "spell it out" and make the page easy to find in a search. Happy to subedit. If neither of you would do this, I can "try" but someone would need to sign off that the untested work was "good enough".

Naish88, 2015-11-23:
Wiki needs to be more clear... I keep getting messages form users asking me about dual battery setup. I'm happy to help making diagrams if needed.

hamishwillee, 2015-11-23:
Hi @Naish88 

If you can create diagrams and information on the setting for this - in MS word or direct on the wiki, I can subedit, get it reviewed, and post.

Thanks!

Naish88, 2015-11-25:
@hamishwillee 
I started writing the doc for the dual battery setup, however I decided to try it myself before I write about it.  I discovered that there are some issues.  Current of the second battery is not displayed, because there is no mavlink message for it  ( as pointed out by Tridge). I will finish to write the doc anyway, and I will see with Tridge if we can make some changes.

hamishwillee, 2015-11-25:
@Naish88 THanks for letting me know.  How useful is updating the wiki if you can't get second battery information? If the answer is "very" then great - please update but we just need to make clear the limitations and provide link to the issue where we track getting this fixed. Alternatively, if this isn't all

Ancistrus01, 2016-01-08:
Hi Guys! 
First think I wanna say is that I want to thanks all developers for their perfect work!
Then I wanna ask how does it looks like with dual battery monitoring?
I am building huge Octo with dual GensAce 16000mAh 6S battery. I want to monitor both batteries independently using two 200A hall sensors. 
So I would be really grateful if you could implement dual battery monitoring to ArduCopter code... :)

iskess, 2016-03-23:
I want to monitor the current of my Quadplane motors in addition to my forward flight motor.  Any progress on this?

triolus, 2016-06-17:
Would it be possible to sent second battery info from the x8r accessory sensors? It would not require an additional plug on the pixhawk that way. Need a 2nd battery monitor for my camera battery.

magicrub, 2016-06-17:
Battery 2 info has been fixed in plane v3.6.0.

hamishwillee, 2016-06-19:
@magicrub In what way has it been fixed? This post originally existed because there are no instructions on setup or config of the second battery.

iskess, 2016-06-19:
Does the batt2 current get sent via mavlink?

hamishwillee, 2016-06-19:
@iskess I can only assume so, because what else would "support" for battery 2 mean?

magicrub, 2016-06-20:
Battery 2 is just like battery 1 now, in v3.6.0. Mavlink and data flash logging wise.

camow7, 2016-06-21:
@magicrub is second battery monitoring setup and working yet? When using a pixhawk I'm using the 3.3V ADC to monitor voltage and current. When I set the Volt and current pin params to 13 and 14 respectively I get no readings? I'm also confused that in the comments for the params it only expects values from 0-13?

magicrub, 2016-06-21:
The comments in the param need to be updated. I can do that

triolus, 2016-07-29:
What about updating arducopter for quads. Can a second brick be added to pins 13/14 and get a battery 2 reading? Also can frsky telemetry voltage and amps be used as a second battery too?

thanks,
Charles C Spencer III

triolus, 2016-07-29:
If I route 2nd brick through pin 12/13, will there be 1. an adjustment set for the 2nd battery in the setup, and 2. will it show up on the gcs flight screen? 3. Will it be sent out via mavlink to an osd?

Thanks,
Charles C. Spencer iii

triolus, 2016-08-12:
If I route 2nd brick through pin 12/13, will there be 1. an adjustment set for the 2nd battery in the setup, and 2. will it show up on the gcs flight screen? 3. Will it be sent out via mavlink to an osd? What I mean to say is: When I go to set up the battery 2 current/voltage, the ground control has 12 and 13, not 13 and 14. It has added a bit of confusion. To note pin 12 is on adc 6.6v.........

Thanks,
 triolus

lincolngerken, 2016-10-27:
@triolus You're right, it makes sense. Any news?

wildtypitch, 2017-12-13:
Over a year old and still not working? Pl;ugged in my second mauch power module today and a little put off by the fact that I get 80amps on battery monitor 2 plugged in or not using a12 and a13 as suggested by the pulldown.

gmorph, 2017-12-15:
I've a project that requires me to use the second battery monitor function.  That should happen around new year and I hope to have the documentation updated at the same time.

ACPUK, 2018-07-17:
MP is setting Battery Monitor 2 current pin to 12 for Pixhawk 2.1 
Should be pin 14

brunoolivieri, 2019-07-15:
The arducopter wiki does have instructions to use a second battery monitor: http://ardupilot.org/copter/docs/common-3dr-power-module.html.

Indeed it is a "common" page displayed also displayed in Rover and Plane.

This information is enough to close the issue?

auturgy, 2019-07-16:
we can probably close this specific issue, although the additional info only covers PH2.1 for the second battery.
Battery monitor in general requires a significant wiki update - much has changed.  @tridge @WickedShell

rmackay9, 2019-07-16:
ok closing..

closed 2019-07-16 by rmackay9