ArduPilot/ardupilot_wiki #156: Setting up SITL on Windows sim_vehicle.sh

ianstankosh, 2015-09-16:
I'm using http://dev.ardupilot.com/wiki/sitl-native-on-windows/ to set up SITL on my computer. I have gotten to the section "Running SITL and MAVProxy." I have followed all of the instructions up to this point. When I type in "sim_vehicle.sh -j4 --map" Cygwin tells me the command is not found. I don't know what to do. Please advise.

hamishwillee, 2015-09-16:
G'day - you're definitely in the ~/ardupilot/ArduPlane,  ~/ardupilot/ArduCopter or  ~/ardupilot/AMPRover2 directory when you call this?

ianstankosh, 2015-09-17:
yes i am. I had a little trouble setting up the path to Tool/autotest in a previous step. I'm not sure if that would affect this?

hamishwillee, 2015-09-17:
@ianstankosh I wrote the instructions and they worked for me. However I'm not an expert on this. I will clean my installation and try follow the instructions again in the next few days. In the meantime, suggest you raise this on the [APM forums](http://ardupilot.com/forum/) to broaden the number of people who might be able to help.

ianstankosh, 2015-09-17:
Thank you. I have already posted this issue to the APM forums.

hamishwillee, 2015-09-17:
Where please - then if someone else answers before I get to this I don't waste time.

hamishwillee, 2015-09-17:
@ianstankosh Thinking about this, your error is basically "not found" which means that Cygwin environment doesn't know where to look for sim_vehicle.sh. I'm betting that you missed this line in the instructions:

3. Add the following line to the end of .bashrc

   ```
export PATH=$PATH:$HOME/ardupilot/Tools/autotest
```

hamishwillee, 2015-09-17:
OK I am getting two errors:

1. **sim_vehicle.sh** is not found because for some reason the .bashrc is not being called. You can add it by calling `export PATH=$PATH:$HOME/ardupilot/Tools/autotest` inside the cygwin terminal directly.

2. When you can call **sim_vehicle.sh** I still get an error:

    ```
hamis_000@XPS12ultra ~/ardupilot/ArduCopter
$ sim_vehicle.sh -j4 --map
~/ardupilot/ArduCopter ~/ardupilot/ArduCopter
Building sitl
make: *** No rule to make target '/tmp/ArduCopter.build/compat.o', needed by '/tmp/ArduCopter.build/ArduCopter.elf'.  Stop.

make: *** No rule to make target '/tmp/ArduCopter.build/compat.o', needed by '/tmp/ArduCopter.build/ArduCopter.elf'.  Stop.
/cygdrive/c/Users/hamis_000/Desktop/%HOMEPATH%/ardupilot/Tools/autotest/sim_vehicle.sh: Build failed
```
   I played with various versions of the cygwin tools but am a bit stuck.

ianstankosh, 2015-09-17:
My post in the APM forums is at the following link:
http://ardupilot.com/forum/viewtopic.php?f=71&t=13763&p=0&e=0&sid=1d1667fcfb94a739ac5731f60ed45349

Those are the problems I am encountering too. Last night I physically found
the .bashrc file in windows explorer. I played around with it a little and
eventually got MAVProxy to run with the map and everything. The only
problem was that the simulated drone was not responding to any of my
commands. I tried to do it again later and now I can't get it to work
anymore. I'm going to keep playing around with it. Please let me know if
you figure out a solution.

hamishwillee, 2015-09-17:
>  I played around with it a little and eventually got MAVProxy to run with the map and everything.

YOu're doing bettter than me- I can't even get .bashrc to load (though as above, I can separately add the path for sim_vehicle.sh). Did you do anything to get it to load, and if so, what?

It looks like cygwin might be messing with my Python installation too.

hamishwillee, 2015-09-17:
PS This is not the only way to get SITL working on Windows. Check out the Vagrant instructions in dev wiki sidebar too.

ericjswenson15, 2015-09-23:
export PATH=$PATH:$HOME/ardupilot/Tools/autotest

That recommendation worked for me, thank you.

hamishwillee, 2015-09-25:
So the setup "used to work for me". What I think happened is that following installation of Windows 10 (or possibly something else) the Windows environment variable %HOMEPATH% stopped expanding properly in Cygwin (and Python). When this happened my homepath wasn't found anymore, which meant that .bashrc no longer loaded and all my existing setup "disappeared". 

This wasn't obvious at the time, but when I reinstalled Cygwin for testing, I discovered that my home directory had changed to ``C:\Users\hamis_000\Desktop\%HOMEPATH%`` (yes, actually a directory named %HOMEPATH% was created :-(. You can add the ``export PATH=$PATH:$HOME/ardupilot/Tools/autotest`` to .bashrc in that directory, but building the sources doesn't work because make chokes on that directory path.

My solution was to rename the %HOMEPATH% directory to *cygwinhome*. Then I updated **c:/cygwin/cygwin.bat** as shown:

```
@echo off
SETLOCAL
set HOME=C:\Users\hamis_000\Desktop\cygwinhome
C:
chdir C:\cygwin\bin

bash --login -i
```
I can no longer launch Cygwin from the desktop shortcut but if I launch the BAT it sets up HOME correctly and the build works as described in the guide!

The right solution would be of course to get my %HOMEPATH% working properly, but this appears to be a Windows 10 issue that no one much has run into so no ideas how to fix. 

@ianstankosh hope this is enough to help you resolve the problem. Was your issue similarly on Windows 10 and related to the cygwin home directory? 
You can check in cygwin terminal by seeing what you get if you enter each of the following lines:

```
echo $HOME
echo $HOMEPATH
printenv
```

hamishwillee, 2015-09-25:
FYI, I don't want to clutter up the main document for a problem that is likely affecting only a few users. So for the doc I've added a [troubleshooting section](http://dev.ardupilot.com/wiki/sitl-native-on-windows/#troubleshooting) (which just links back to this thread) and I've added a "tip" to the section on [setting up paths](http://dev.ardupilot.com/wiki/sitl-native-on-windows/#set_up_directoriespaths_in_cygwin) which I hope will give users clues on the likely cause of problems by putting the likely error here.

jbhargavr, 2015-10-14:
hey,
could you please explain me what the following sentence mean-
2.Navigate the file system to the home directory and open the .bashrc files (e.g. C:\cygwin\home\user_name\.bashrc.

hamishwillee, 2015-10-14:
@jbhargavr It means exactly what it says :-)
[image]

There is a file **.bashrc** which gets loaded when you open a bash terminal (i.e. Cygwin console in this case). These instructions explain how to find that .bashrc file, and add a line which adds the location of the /ardupilot/Tools/autotest directory to the PATH for that bash console. Once you have done this the terminal will be able to find the sim_vehicle.sh file, which is located in /ardupilot/Tools/autotest

hamishwillee, 2015-10-14:
I am closing this issue. Ultimately the problem is that either people did not follow the instructions or  Windows has broken with respect to its handling of the path variable. In either case it is not really a fault of the instructions.

What I have done is highlighted in the doc the possible issue and linked to  https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/156#issuecomment-143084698 for possible solutions.

closed 2015-10-14 by hamishwillee

uri200, 2015-12-04:
Hello,
I am trying to build AurduCopter at CygWin and I am getting ArduCopter.elf' failed when i try toi execute make sitl -j4. Any fix for this issue?

C:/cygwin/home/uri/ardupilot/mk/board_native.mk:106: recipe for target '/tmp/ArduCopter.build/ArduCopter.elf' failed
make: *** [/tmp/ArduCopter.build/ArduCopter.elf] Error 1

pjmosquera, 2016-04-29:
@hamishwillee Hello Hamish, I did what the instructions says. But I figured out that the code lines of my .bashrc file is all written with "#". ¿What should I do?

----------------------------------------------------------------------------------------------------------------------

# To the extent possible under law, the author(s) have dedicated all 
# copyright and related and neighboring rights to this software to the 
# public domain worldwide. This software is distributed without any warranty. 
# You should have received a copy of the CC0 Public Domain Dedication along 
# with this software. 
# If not, see <http://creativecommons.org/publicdomain/zero/1.0/>. 

# base-files version 4.2-4

# ~/.bashrc: executed by bash(1) for interactive shells.

# The latest version as installed by the Cygwin Setup program can
# always be found at /etc/defaults/etc/skel/.bashrc

# Modifying /etc/skel/.bashrc directly will prevent
# setup from updating it.

# The copy in your home directory (~/.bashrc) is yours, please
# feel free to customise it to create a shell
# environment to your liking.  If you feel a change
# would be benifitial to all, please feel free to send
# a patch to the cygwin mailing list.

# User dependent .bashrc file

# If not running interactively, don't do anything
[[ "$-" != *i* ]] && return

# Shell Options
#
# See man bash for more options...
#
# Don't wait for job termination notification
# set -o notify
#
# Don't use ^D to exit
# set -o ignoreeof
#
# Use case-insensitive filename globbing
# shopt -s nocaseglob
#
# Make bash append rather than overwrite the history on disk
# shopt -s histappend
#
# When changing directory small typos can be ignored by bash
# for example, cd /vr/lgo/apaache would find /var/log/apache
# shopt -s cdspell

# Completion options
#
# These completion tuning parameters change the default behavior of bash_completion:
#
# Define to access remotely checked-out files over passwordless ssh for CVS
# COMP_CVS_REMOTE=1
#
# Define to avoid stripping description in --option=description of './configure --help'
# COMP_CONFIGURE_HINTS=1
#
# Define to avoid flattening internal contents of tar files
# COMP_TAR_INTERNAL_PATHS=1
#
# Uncomment to turn on programmable completion enhancements.
# Any completions you add in ~/.bash_completion are sourced last.
# [[ -f /etc/bash_completion ]] && . /etc/bash_completion

# History Options
[... 109 lines cut ...]
#   [[ $? -ne 0 ]] && return 1
#   the_new_dir=$(pwd)
# 
#   #
#   # Trim down everything beyond 11th entry
#   popd -n +11 2>/dev/null 1>/dev/null
# 
#   #
#   # Remove any other occurence of this dir, skipping the top of the stack
#   for ((cnt=1; cnt <= 10; cnt++)); do
#     x2=$(dirs +${cnt} 2>/dev/null)
#     [[ $? -ne 0 ]] && return 0
#     [[ ${x2:0:1} == '~' ]] && x2="${HOME}${x2:1}"
#     if [[ "${x2}" == "${the_new_dir}" ]]; then
#       popd -n +$cnt 2>/dev/null 1>/dev/null
#       cnt=cnt-1
#     fi
#   done
# 
#   return 0
# }
# 
# alias cd=cd_func

export PATH=$PATH:$HOME/ardupilot/Tools/autotest

pjmosquera, 2016-04-29:
@hamishwillee Hello Hamish, I did what the instructions says. But I figured out that the code lines of my .bashrc file is all written with "#". ¿What should I do?

`# To the extent possible under law, the author(s) have dedicated all 
# copyright and related and neighboring rights to this software to the 
# public domain worldwide. This software is distributed without any warranty. 
# You should have received a copy of the CC0 Public Domain Dedication along 
# with this software. 
# If not, see <http://creativecommons.org/publicdomain/zero/1.0/>. 

# base-files version 4.2-4

# ~/.bashrc: executed by bash(1) for interactive shells.

# The latest version as installed by the Cygwin Setup program can
# always be found at /etc/defaults/etc/skel/.bashrc

# Modifying /etc/skel/.bashrc directly will prevent
# setup from updating it.

# The copy in your home directory (~/.bashrc) is yours, please
# feel free to customise it to create a shell
# environment to your liking.  If you feel a change
# would be benifitial to all, please feel free to send
# a patch to the cygwin mailing list.

# User dependent .bashrc file

# If not running interactively, don't do anything
[[ "$-" != *i* ]] && return

# Shell Options
#
# See man bash for more options...
#
# Don't wait for job termination notification
# set -o notify
#
# Don't use ^D to exit
# set -o ignoreeof
#
# Use case-insensitive filename globbing
# shopt -s nocaseglob
#
# Make bash append rather than overwrite the history on disk
# shopt -s histappend
#
# When changing directory small typos can be ignored by bash
# for example, cd /vr/lgo/apaache would find /var/log/apache
# shopt -s cdspell

# Completion options
#
# These completion tuning parameters change the default behavior of bash_completion:
#
# Define to access remotely checked-out files over passwordless ssh for CVS
# COMP_CVS_REMOTE=1
#
# Define to avoid stripping description in --option=description of './configure --help'
# COMP_CONFIGURE_HINTS=1
#
# Define to avoid flattening internal contents of tar files
# COMP_TAR_INTERNAL_PATHS=1
#
# Uncomment to turn on programmable completion enhancements.
# Any completions you add in ~/.bash_completion are sourced last.
# [[ -f /etc/bash_completion ]] && . /etc/bash_completion

# History Options
#
# Don't put duplicate lines in the history.
[... 106 lines cut ...]
#   pushd "${the_new_dir}" > /dev/null
#   [[ $? -ne 0 ]] && return 1
#   the_new_dir=$(pwd)
# 
#   #
#   # Trim down everything beyond 11th entry
#   popd -n +11 2>/dev/null 1>/dev/null
# 
#   #
#   # Remove any other occurence of this dir, skipping the top of the stack
#   for ((cnt=1; cnt <= 10; cnt++)); do
#     x2=$(dirs +${cnt} 2>/dev/null)
#     [[ $? -ne 0 ]] && return 0
#     [[ ${x2:0:1} == '~' ]] && x2="${HOME}${x2:1}"
#     if [[ "${x2}" == "${the_new_dir}" ]]; then
#       popd -n +$cnt 2>/dev/null 1>/dev/null
#       cnt=cnt-1
#     fi
#   done
# 
#   return 0
# }
# 
# alias cd=cd_func
export PATH=$PATH:$HOME/ardupilot/Tools/autotest`

pjmosquera, 2016-04-29:
@hamishwillee Hello Hamish. I did what the instructions says. But I figured out that the code lines of my .bashrc file is all written with "#". ¿What should I do?

magicrub, 2016-04-29:
Woah, you have an alias for "cd"? That can't be good.

pjmosquera, 2016-04-29:
woah woah woah. I´m going to reinstall this. maybe I put this trying to find a solution. But what about wint the numerals (#). Do I have to write in this way?:
# export PATH=$PATH:$HOME/ardupilot/Tools/autotest  ??

pjmosquera, 2016-04-29:
woah woah woah. I´m going to reinstall this. maybe I put this trying to find a solution. But what about wint the numerals (#). Do I have to write in this way?:

# `export PATH=$PATH:$HOME/ardupilot/Tools/autotest `??

pjmosquera, 2016-04-29:
woah woah woah. I´m going to reinstall this. maybe I put this trying to find a solution. But what about wint the numerals (#). Do I have to write in this way?:

## export PATH=$PATH:$HOME/ardupilot/Tools/autotest ??

pjmosquera, 2016-04-29:
woah woah woah. I´m going to reinstall this. maybe I put this trying to find a solution. But what about wint the numerals (#). Do I have to write in this way?:

`( numeral ) export PATH=$PATH:$HOME/ardupilot/Tools/autotest ??`

>

pjmosquera, 2016-05-04:
@magicrub I'm in Windows 7 and I have reinstalled Cygwin. For some reason adding the line to the end of .bashrc file `export PATH=$PATH:$HOME/ardupilot/Tools/autotest` (with or without #, nobody explained me this point, for me all the code in coments has not logic) is not working for me. Alias for "cd" is a part of the original .bashrc code. My orthodox solution is to open the Cygwin Terminal, and in my home directory enter `export PATH=$PATH:$HOME/ardupilot/Tools/autotest` each time before calling sim_vehicle.sh.

tsimmons15, 2016-08-03:
@pjmosquera The # is a comment. The shell interpreter should ignore lines beginning with a #. This doesn't hold true for all of them, but the default Linux shells (bash/dash/etc.) should interpret it as a comment.

pjmosquera, 2016-08-03:
@tsimmons15 so, why I had to add a line "export PATH = etc etc" to a .bashrc file that is all in comment? By the way. Why codes of this file is all in comments?  Do I make myself clear ?

hamishwillee, 2016-08-04:
@pjmosquera If every line in your .bashrc file is a comment then it isn't doing anything. That is probably because on cygwin it is provided with lots of stuff you can enable if you want ... by removing the comments. 

The suggestion is to add "export PATH = etc etc" to a bashrc NOT in comments, so that this line is enabled. Essentially this means that when the .bashrc file is loaded only that line will be executed.

I wish people would start new questions, ideally on the discussion boards, rather than hijacking this very long and old issue.

jayjagtap, 2017-10-02:
I have the following output when I execute make sitl -j4
I have followed the installation steps as in 
http://ardupilot.org/dev/docs/sitl-native-on-windows.html

hp@LAPTOP-59TH2EJC /cygdrive/c/Users/hp/ardupilot/ArduCopter
$ make sitl -j4
Checking modules
echo "Generating MAVLink headers..."
Generating MAVLink headers...
Generating UAVCAN headers...
PYTHONPATH=/cygdrive/c/Users/hp/ardupilot/modules/uavcan/libuavcan/ python /cygd rive/c/Users/hp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc "/cygdrive/c/Users/hp/ardupilot/modules/uavcan/dsdl/uavcan" -O"/cygdrive/c/User s/hp/ardupilot/modules/uavcan/libuavcan/include/dsdlc_generated/"

goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /cygdrive/c/Users/hp/ardupil ot/modules/mavlink/
Generating C code using mavgen.py located at /cygdrive/c/Users/hp/ardupilot/modu les/mavlink/
PYTHONPATH=/cygdrive/c/Users/hp/ardupilot/modules/mavlink/ python /cygdrive/c/Us ers/hp/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-prot ocol=2.0 --output=/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v2 .0 /cygdrive/c/Users/hp/ardupilot/modules/mavlink/message_definitions/v1.0/ardup ilotmega.xml; if [ $? -le 0 -o $? -gt 128 ]; then echo "mavgen: success"; exit 0 ; else echo "mavgen: failed"; exit 1; fi
Traceback (most recent call last):
File "/cygdrive/c/Users/hp/ardupilot/modules/mavlink//pymavlink/tools/mavgen.p y", line 16, in 
from pymavlink.generator import mavgen
File "/cygdrive/c/Users/hp/ardupilot/modules/mavlink/pymavlink/generator/mavge n.py", line 12, in 
from future import standard_library
ImportError: No module named future
mavgen: failed
make: *** [C:/Users/hp/ardupilot/mk/mavgen.mk:12: /tmp/ArduCopter.build/librarie s/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega/mavlink.h] Error 1
make: *** Waiting for unfinished jobs....
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HA L_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="42add82e" -I/cyg drive/c/Users/hp/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSIO N=2 -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE -DHAVE_ENDIAN_H -D HAVE_BYTESWAP_H -DUAVCAN_CPP_VERSION=UAVCAN_CPP03 -DUAVCAN_NO_ASSERTIONS -DUAVCA N_NULLPTR=nullptr -I/cygdrive/c/Users/hp/ardupilot/modules/uavcan/libuavcan/incl ude -I/cygdrive/c/Users/hp/ardupilot/modules/uavcan/libuavcan/include/dsdlc_gene rated

Please tell me what is wrong. All above steps has been successfully executed.

peterbarker, 2017-10-03:
On Mon, 2 Oct 2017, jayjagtap wrote:

> I have the following output when I execute make sitl -j4
> I have followed the installation steps as in
> http://ardupilot.org/dev/docs/sitl-native-on-windows.html

> Please tell me what is wrong. All above steps has been successfully
> executed.

Are you *sure* these steps ran successfully?

http://ardupilot.org/dev/docs/sitl-native-on-windows.html#install-required-python-packages

ejhaze, 2017-11-15:
I followed all the steps, but I seemed to unlock the SITL with: 
./sim_vehicle.py -v ArduCopter

Of course, change "copter" to your project.

nagendra092918, 2019-02-11:
Traceback (most recent call last):
  File "/home/nagendra/ardupilot/Tools/autotest/sim_vehicle.py", line 25, in <module>
    from MAVProxy.modules.lib import mp_util
ImportError: No module named MAVProxy.modules.lib

i got this error when i am installing sitl on windows .please help  me