ArduPilot/ardupilot_wiki #157: Pixhawk overview conflicting documentation

Georacer, 2015-09-17:
Hello again!

In the [Pixhawk overview page](http://plane.ardupilot.com/wiki/common-pixhawk-overview/#connecting_and_disconnecting_df13_connectors) we can read below the bare board image that
> [...]and Serial 3 (GPS) Pins: 6 = GND, 5 = RTS, 4 = CTS, 3 = RX,  2 = TX, 1 = 5V.
However, below it reads that pins 4 and 5 are a CAN bus for the GPS port.

I think the second is correct, right?

rmackay9, 2015-09-17:
Yes, I think @Georacer is correct.  We don't have CTS, RTS pins on Serial 3.  I actually didn't know that those pins were for the 2nd CAN bus (aka "CAN2") but in any case we should remove "Serial 3" from the list that appears just below "For all connectors pin 1 is on the right in the above image."

hamishwillee, 2015-09-18:
So would I be correct in assuming that ALL I need to do here is remove the Serial 3 from the list? (done - [diff here](http://planner.ardupilot.com/wp-admin/revision.php?from=5114&to=5132))

That makes sense as that is the only place we say the "wrong thing". OK for me to close this?

rmackay9, 2015-09-18:
yup, sounds good to me.

closed 2015-09-18 by hamishwillee