ArduPilot/ardupilot_wiki #160: common-FrSky Telemetry

lvale, 2015-09-23:
Hi. 
As the tested versions are using OpenTX 2.0.x, and the telemetry system has changed radically on OpenTX 2.1.x I believe a note warning that this page is tested using ONLY versions of OpenTX 2.0.x

There are already questions and "misdirections" on forae about this issue, so the note would be helpful.

[common-FrSky Telemetry](http://copter.ardupilot.com/wiki/common-frsky-telemetry/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

hamishwillee, 2015-09-23:
Thanks Luis. I've submitted change (in review) to the wiki editors group. The note says "These instructions were tested using OpenTX 2.0.x (only), and have not yet been verified on OpenTX 2.1.x."

It would be great if we could verify the changes against the new version. Is that something you can do, or is there someone you can recommend we ask to get this updated?

lvale, 2015-09-23:
I can't test the solution described for lack of hardware :) Perhaps the best person would be Matthias 
https://github.com/badzz

hamishwillee, 2015-09-24:
@badzz - didn't answer my last message. If you have a personal relationship with him could you point him here?

lvale, 2015-09-24:
sorry, no. only met him here...Perhaps Rob can help, this was discussed a few weeks ago on a dev call and I remember Rob saying he had the necessary hw.

lvale, 2015-09-24:
Some more corrections, that a user warned me:

on the TELEM2 line should read:

Telem2: SERIAL2_BAUD, SERIAL2_PROTOCOL, BRD_SER2_RTSCTS

Set-up through the Mission Planner
The telemetry port is configured using parameters, which you can edit though your GCS. Using Mission Planner open the Config/Tuning | Full Parameter List page.

The parameters to change for each port are listed below. These are documented with the parameter in Mission Planner and in the Parameter documentation on each wiki:

Telem1: SERIAL1_BAUD, SERIAL1_PROTOCOL, BRD_SER1_RTSCTS
Telem2: SERIAL2_BAUD, SERIAL1_PROTOCOL, BRD_SER1_RTSCTS

hamishwillee, 2015-09-24:
Hi @lvale - you mean Robert Lefebvre?

W.r.t. above issue, thanks for pointing it out. This was not on the FrSky page, but I fixed this on [Telemetry Port Setup for Pixhawk and PX4](http://planner.ardupilot.com/wiki/common-telemetry-port-setup-for-apm-px4-and-pixhawk/#set-up_through_the_mission_planner) - presumably that is where you meant?

lvale, 2015-09-24:
Yes, correct page. I have Rob in Mumble and already reminded him, but he is kind of busy with Heli :)

hamishwillee, 2015-09-24:
@R-Lefebvre Apparently [common-FrSky Telemetry](http://copter.ardupilot.com/wiki/common-frsky-telemetry/) is accurate for  OpenTX 2.0.x (only) but may have issues for OpenTX 2.1.x. 

It would be good to document what needs to change for OpenTX 2.1.x. I understand that you have a setup that can be used to test this. I appreciate you are busy - do you think you might be able to check this at some point, and if so, "roughly" when?

mpjf01, 2015-09-24:
I have  a complete set of FRSKY D Series telemetry sensors, a number of the X series, several APMs and Pixhawks and one Taranis running 2.0.17 and another running 2.1.2. Let me know what (if anything) I can do to help.

lvale, 2015-09-24:
@mpjf01 I believe that what would be useful is to replicate (and document) the steps provided at the wiki page http://copter.ardupilot.com/wiki/common-frsky-telemetry/ and validate them with either 2.0.x and 2.1.x versions of OpenTX, and also with Copter 3.2.1 and with the latest 3.3 (RC12 now) to ensure the steps described work with all the combinations of Copter firmware and OpenTX.

mpjf01, 2015-09-24:
What is the non-annotated device sitting between the FUL-1 and the X4R receiver in the X Receiver Setup section? I have all the other components.

badzz, 2015-09-24:
I will test on opentx 2.1 but last time i tried it was working Ok.
Sorry for the lack of communication

hamishwillee, 2015-09-24:
@badzz Thanks very much. What I should have done is asked Craig to help me get in touch with you - appreciate you're busy and probably getting a lot of email noise.

Can you answer @mpjf01 [question above](https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/160#issuecomment-142931537)?

More generally, @lvale thinks that incompatibilities are possibly going to occur across Copter/OpenTX versions. Does this make sense? Either way it is probably worth identifying exactly versions this was verified against so we have a good baseline.

badzz, 2015-09-26:
@mpjf01 : it is a "SPC cable"

badzz, 2015-09-26:
I did a check with latest opentx 2.1.3 and latest beta AC 3.3rc12  and it works fine.
SERIAL2_BAUD and SERIAL1_BAUD are not necessary for frsky telem as we force the speed of serial for frsky as it is a static value for D-Port (9600) and S-port (57600)
Only thing with opentx 2.1 is to know what is t1 ( flight mode ) and t2 (number of sats) as they are both called Temp, but switching mode tells you that quickly.
The rest of opentx 2.1 issue is that you loose your opentx 2.0 config and need to "discover" your  sensors but that's just opentx new way of doing things.

hamishwillee, 2015-09-28:
@badzz Thanks very much. I have made some [modifications](http://planner.ardupilot.com/wp-admin/revision.php?from=5142&to=5144) in line with comments. Can you please sanity check them?

1. [X-Receiver setup image](http://planner.ardupilot.com/wiki/common-frsky-telemetry/#x-receiver_setup) now includes SPC Cable label.
1. [Additional note added](http://planner.ardupilot.com/wiki/common-frsky-telemetry/#softwarehardware_solution) stating version tested:

   > These instructions were written and tested for OpenTX 2.0.x. They have been re-validated against OpenTx 2.1.3 and AC 3.3rc12.
1. Added [tip to telemetry setup](http://planner.ardupilot.com/wiki/common-frsky-telemetry/#frsky_telemetry_setup_in_mission_planner) noting that the baud rate setting is not needed
1. Added note in [transmitter setup](http://planner.ardupilot.com/wiki/common-frsky-telemetry/#transmitter_set-up)

   > If upgrading to OpenTx 2.1 you will need to replace your OpenTx 2.0 configuration and “discover” your sensors. There are other minor “oddities” – for example T1 (flight mode) and Tt2 (number of sats) are both called TEMP (switching mode helps you identify which is which).

In particular is the last note correct? I have tried to paraphrase.

Thanks very much!

mpjf01, 2015-09-28:
Thanks for the info. Which field is used for HDOP (was A2 in one incarnation)?

closed 2015-10-13 by lvale

hamishwillee, 2015-10-14:
@mpjf01 Sorry - I missed your [comment](https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/160#issuecomment-143613949). 

The right place for suggestions about improving ardupilot itself is the ardupilot issue tracker on github - https://github.com/diydrones/ardupilot (The tracker is used for wiki improvements and issues)