ArduPilot/ardupilot_wiki #161: Traditional Heli Documentation Updates

Jman841, 2015-09-24:
The Traditional Heli Documentation on the Wiki, and, description of Parameters is severely outdated, or, does not exist.

Trying to guess what different parameters do is quite challenge and I can't seem to find any resources that explain it like they do for Arduplane or Arducopter.

hamishwillee, 2015-09-24:
True. @R-Lefebvre - would appreciate your advice on how to fix this.

hamishwillee, 2015-09-24:
@Jman841 What parameters information specifically have you found out of date - and what are you expecting? Can you identify the document/parameters that you found to be incorrect. What do you expect to see - something like http://copter.ardupilot.com/wiki/arducopter-parameters/ ? 

We can certainly improve these documents, but would be really helpful to start with specific problem areas.

hamishwillee, 2015-09-24:
Robert has just told me we plan to update this whole section for AC3.3 (the release nearing final stability now). I'll keep this open for your advice and sugestions.

Jman841, 2015-09-25:
Hello Hamishwillee, I will go through the parameters tomorrow when I have my heli handi to look at them all, but, the majority of parameters do not have any description or recommended values in Mission planner.

The feed forward values and the critical headspeed values are some I can think of off the top of my head. For feed forward, understanding what too much or too litle of it looks like, and, a general value range would be greatly appreciated. For the critical headspeed value, Rob recently explained this to me ,but ,when setting up my heli for the first time I did not know what this was used for and I had issues with it.

For the Wiki, it is severely outdated. One part of it even states that you can not do flips due to the earth plane of the flight controller which was resolved a couple years ago.

I have a few years background with 3D Heli's, extensive use of arduplane and arducopter, and had trouble figuring everything out with the current documentation. I could not imagine trying to set it up as a beginner with no background in Heli's or no other Ardupilot background. 

I will gladly help write the wiki updates, but, I need to fully understand it first.

hamishwillee, 2015-09-25:
Thanks very much. Robert *does* fully understand the current heli and needed updates. However as a power user you will make a very good test case for the updates. Would really appreciate your review and testing as the updates go in (in the coming months)

Jman841, 2015-09-25:
I will do the best I can, I compiled robs latest work in progress firmware from his branch to test this weekend, just waiting for the OK from him. We plan to build some UAV helicopters in the near future and will be hiring a software engineer to help with the development of Ardupilot as we find it is by far the best option right now for an autopilot system. My personal skill is limited to posting issues, but, hopefully soon we will be able to contribute more

CraigElder, 2015-09-28:
@Jman841 
>>>the majority of parameters do not have any description or recommended values in Mission planner.
Parameter descriptions that appear in the planners come from the Master branch of the code not code that is in development.  If you parameter descriptions you'll have to retrieve them directly from the branch / code that is in development.

closed 2015-09-28 by CraigElder