ArduPilot/ardupilot_wiki #166: Communicating with Raspberry Pi via MAVLink

ruderik1, 2015-09-28:
[Communicating with Raspberry Pi via MAVLink](http://dev.ardupilot.com/wiki/raspberry-pi-via-mavlink/) 

In the chapter "Install the required packages on the Raspberry Pi" it would help to mention the installation of additional dependencies;

`sudo apt-get install python-dev`
`sudo apt-get install libxml2-dev`
`sudo apt-get install libxslt-dev`

before next lines can be executed:
`sudo pip install pymavlink`
`sudo pip install mavproxy`

Thanks

hamishwillee, 2015-09-28:
Hi @ruderik1 

I've added python-dev as that is certainly a dependency. Are you absolutely sure about libxml2-dev and libxslt-dev - I didn't need these for ubuntu and haven't heard of the need elsewhere?

ruderik1, 2015-09-29:
Hi @hamishwillee,

I'm sorry, should have mentioned the platform; it's raspbian (on a rpi b). I don't have the logs at hand, but the installer was complaining about missing libxslt and libxml2. Those were actually installed, but not the dev versions, after installing them, everything was working fine.
Note that i'm by no means an experienced rpi user, that's why i was using the wiki ;)

best regards,
Ruud

hamishwillee, 2015-09-29:
Hi Ruud, 

Fair enough to assume Raspian given the doc you referenced!

I'm seeking confirmation, but I'll add these dependencies even if I don't get it (within a few days) because I believe it won't do significant harm to install them if not present.

Thanks very much

Hamish

ruderik1, 2015-09-29:
Hi,

That's right. Thanks for the support.

hamishwillee, 2015-09-30:
No - thank you for talking the time to let us know about a problem. I've made the update now. I'll leave this open for a bit longer though in case I get feedback.

closed 2015-10-14 by hamishwillee