ArduPilot/ardupilot_wiki #167: common-Camera Shutter Configuration in Mission Planner

mauro18421, 2015-09-30:
[common-Camera Shutter Configuration in Mission Planner](http://planner.ardupilot.com/wiki/common-camera-shutter-with-servo/) - Mission Planner Reset to "0" (Servo Output) the CAM_TRIG_TYPE param.

Every time I change the trigger duration or any other parameter on the Camera Guimbal config page  the CAM_TRIG_TYPE goes to cero and the output of the selected Aux pin start to work in Servo mode. No mater if it was previously set to cero (Servo) or "1" (Relay). So the camera stop working until you manualy set the CAM_TRIG_TYPE to 1.

[image]

mauro18421, 2015-09-30:
I sorry about the image wich I take from the web. It's not a real image from my PC. Im runing the last version of Mission Planner ... I think it's the 3.32.

hamishwillee, 2015-09-30:
@mauro18421 This sounds like a bug in *Mission Planner* rather than the documentation. I suggest you update to the latest version of MP, retest this. If it still occurs, post this as an issue in the [Mission Planner repo here](https://github.com/diydrones/MissionPlanner/issues). The title would be something like "Setting shutter duration resets relay pin to be a servo".

I am closing this, but would appreciate it if you could add the issue you create here in case someone else runs into the same problem. Thanks!

closed 2015-09-30 by hamishwillee