ArduPilot/ardupilot_wiki #170: Improve behaviour of site-specific content

hamishwillee, 2015-10-02:
The site supports wiki-specific content using the [site_Copter], [site_Rover], [site_Plane] tags. For logged-in users the content is rendered on all wikis with the tags displayed. For non-logged in users only the content marked up with the appropriate tag for the wiki is displayed - except on planner wiki, where all content is displayed. There is no way to specify that some content should be displayed on multiple wikis.

The proposal is to support a new shortcode (replacing the old one over time):

* Support multiple-wiki content - so for example ``[site wikis="copter,plane"][/site]`` would display on both Copter and Rover
* Have icon or colour coding to indicate that content is wiki specific for all users
* For logged in users make the non-wiki relevant content less prominent - e.g. by making it faded/partially transparent.

gmorph, 2015-10-07:
Sounds like a great idea.

hamishwillee, 2015-10-07:
Hi @gmorph 

Glad you think so. I've started experimenting on the test wiki using the existing shortcodes to test different markup/design.

[On Planner here](http://planner.ardupilot.tridgell.net/wiki/common-documentation-page-structure-test/#site-specific_content) and [here](http://planner.ardupilot.tridgell.net/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_change_speed) you can see the different site-content colour coded. I haven't marked up text or added an icon as a "key" to colours vs site, but I think it is fairly easy to work out..

On the vehicle wikis the non-site content is much more transparent (and obviously less relevant). See [Plane example](http://plane.ardupilot.tridgell.net/wiki/common-documentation-page-structure-test/) and [Copter example](http://copter.ardupilot.tridgell.net/wiki/common-mavlink-mission-command-messages-mav_cmd/#plane-11). I don't think this works as well. 

My thought is that perhaps we should hide the site-irrelevent sections even for logged-in users because really, if I'm on Copter wiki, even if logged in it isn't useful to me to know that there is hidden copter material present. 

Thoughts?

gmorph, 2015-10-11:
Rover should be green - I like green.
As an editor I do want to see the content for the other Wiki's.  I want to know our information and wording is consistent across the wikis.  Hiding it from me isn't going to work.  I understand your concern about editors thinking they are seeing the correct final page when they aren't but each editor should be "previewing" their edited page before submitting it.  That preview should show them what the end user is going to see so they can confirm it for correctness.

Thanks, Grant.

hamishwillee, 2015-10-11:
@gmorph The colours are based on the "theme colour" for each wiki - already pre-determined by your wiki logo. That's brown on Rover :-) . If you want to change colours you need to duel @rmackay9 at dawn for the "right to be green" and arrange new logos.

Absolutely agree that when editing you need to see all the wikis. This is not an issue though because editing is done in the common wiki (Mission Planner), where content for all wikis is shown. 

On other wikis I replace the non-current-wiki-content with the above placeholder. As an editor you're not editing, so IMO you don't need it for comparison purposes. For me it is much easier to use the content when logged in if I can't see all the content for other wikis. Does that clarify/make sense, or do you (still?) think editors need to see all content for all wikis on all wikis?

gmorph, 2015-10-12:
Ok - I've clarified with Hamish - he said (so I have it in writing ;-)

[11:20:12 AM] Hamish Willee: Yes. On common wiki what a user and editor sees is the same - all the content for all the vehicle wikis. All that changes from currently on common wiki is that in future vehicle-specific content is shown with colour coding.
[11:20:19 AM] Hamish Willee: (and a heading)

Which is fine by me.  Thanks Hamish!

hamishwillee, 2015-10-12:
Excellent. Thanks

I have rolled this out to the live server and started updating *some* of the text. I am using [this search](https://github.com/diydrones/ardupilot_wordpress_sources/search?utf8=%E2%9C%93&q=site_Copter) to find unconverted instances.

This is a work in progress. I'll close it when all cases have been addressed (note, the old shortcode is still active, and it may be that I leave it in place if I discover cases that this new shortcode can't fix.)

hamishwillee, 2015-10-13:
Have now updated all sources to use new shortcode (though old one is still available). Also updated https://github.com/diydrones/ardupilot_wordpress_sources with latest versions.

closed 2015-10-13 by hamishwillee