ArduPilot/ardupilot_wiki #171: Building ArduPilot with Eclipse

sunzhon, 2015-10-04:
When I build the Arducopter in eclipse ,the console echo the follow error, 
"../mk/environ.mk:44: *** ERROR: cannot determine sketchbook location - please specify on the commandline with SKETCHBOOK=<path>.  Stop.",I did not know what happen to the solution.

hamishwillee, 2015-10-06:
The best place to raise technical questions is the [APM forums](http://ardupilot.com/forum/). This repot is for feedback about specific documents. When/if you post you should leave the link to the document you have an issue with and also explain exactly what host/tools you used if they differ.

That said, I have just retested the instructions on Windows 10. 

The PX4/Pixhawk instructions work:

* [Building ArduPilot for Pixhawk/PX4 on Windows with Make](http://dev.ardupilot.com/wiki/building-px4-with-make/)
* [Building for Pixhawk/PX4 using Eclipse on Windows](http://dev.ardupilot.com/wiki/editing-the-code-with-eclipse/)

The instructions that build APM2 using make via Cygwin (both on command line and eclipse) result in a number of errors. I have not been able to get these to work on Windows 10:

* [Building ArduPilot with Eclipse]{http://dev.ardupilot.com/wiki/building-with-eclipse/) - Windows/APM2
* [Building ArduPilot for APM2.x with Make](http://dev.ardupilot.com/wiki/building_with_make/#windows) - the "Windows" section

I have however got APM to build by following the Pixhawk links above and running the ``make apm2`` commands in the PX4 console (after adding the Arduino toolchain). I will update these instructions today (the idea for this came from @rmackay9 [here](https://groups.google.com/d/msg/drones-discuss/YooljrxL36A/Ntm7qw55740J).

Note that APM2.x will no longer build from master for any platform. If you need to build for this target then use a branch like ``ArduCopter-3.2``.

hamishwillee, 2015-10-07:
OK, I've updated the topics on building for APM2.x and using Eclipse. Detail below. This is now closed.

Changes Made:

* (new) [Building ArduPilot for APM2.x on Windows with Make](http://dev.ardupilot.com/wiki/building-ardupilot-for-apm2-x-on-windows-with-make/). This has disclaimers that it is "not the recommended approach". It uses the PX4 toolchain. Added to sidebar.
* (updated) [Building ArduPilot for APM2.x on Windows with Arduino](http://dev.ardupilot.com/wiki/building-ardupilot-with-arduino-windows/) ([diff](http://dev.ardupilot.com/wp-admin/revision.php?from=5551&to=6245)) - Added tip noting that this article has the recommended approach for building for APM2.x,  that we can no longer build off master, and nothing the last plane build number that supports APM2.
* (new/updated) [Building ArduPilot for APM2.x with Eclipse on Windows](building-apm2-with-eclipse-on-window)  - completely re-wrote to use the PX4 Eclipse (Luna) rather than download eclipse and integrate with cygwin based make. This is much easier. It is a virtual copy of the equivalent Pixhawk eclipse doc (from which it derives). The main difference is the build targets and the fact that you have to create the project from scratch because the project templates are not in the old codelines that are still able to build with APM2.x.
* (updated) [Building ArduPilot for APM2.x with Make](http://dev.ardupilot.com/wiki/building_with_make/#windows) (windows section) ([diff](http://dev.ardupilot.com/wp-admin/revision.php?from=6237&to=6249)) - Changed this to point to new [Building ArduPilot for APM2.x on Windows with Make](http://dev.ardupilot.com/wiki/building-ardupilot-for-apm2-x-on-windows-with-make/) - previously "basic" info on cygwin solution. Also added the standard note about APM support in ArduPilot vehicles.
* (updated) [Building ArduPilot for APM2.x on Windows with Arduino](http://dev.ardupilot.com/wiki/building-ardupilot-with-arduino-windows/) ([diff](http://dev.ardupilot.com/wp-admin/revision.php?from=5551&to=6258))

I didn't want to have the Eclipse docs in both "Building" and "Code Editors/IDEs section" so I moved the eclipse docs to the  Building section. I created a new doc [Eclipse IDE](http://dev.ardupilot.com/wiki/eclipse-ide/) which references them, and which lives in the IDE section.

closed 2015-10-07 by hamishwillee