ArduPilot/ardupilot_wiki #172: Check and update the MAV_CMDs doc in line with updates to code

hamishwillee, 2015-10-05:
[MAVLink Mission Command Messages (MAV_CMD)](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd) appears to be missing some additions to the code. 

Recheck the files referenced in the [how acccurate is the information](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#how_accurate_is_the_information) section and add any missing docs.

hamishwillee, 2015-10-06:
@gmorph Only appear to be new Plane methods - see below ([diff here]()). Can you please sanity check them:

1. [MAV_CMD_DO_AUTOTUNE_ENABLE](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_autotune_enable)
1. [MAV_CMD_NAV_ALTITUDE_WAIT](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_nav_altitude_wait)
1. [MAV_CMD_NAV_LOITER_TO_ALT](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_nav_loiter_to_alt)

I've also linked to MAV_CMD_DO_AUTOTUNE_ENABLE from [Plane Commands in Guided Mode](http://dev.ardupilot.com/wiki/plane-commands-in-guided-mode/#mav_cmds) as this is a guided command too. The others don't appear to be (right?)

There were also a few changes to MAV_CMD_DO_DIGICAM_CONTROL but these are being discussed [here](https://github.com/diydrones/ardupilot/issues/2670#issuecomment-145688973).

I can't see the second two in MP so can't say what the MP heading is. Have marked as ?

@gmorph @rmackay9 Are there any other new and changed commands you think I might have missed in the last few months - both mission and guided?

gmorph, 2015-10-13:
MAV_CMD_DO_AUTOTUNE_ENABLE is a way to set the mode to autotune.  autotune is like FBWA but it monitors the plane performance and adjusts PID values accordingly.  I wouldn't link it to Plane Commands in Guided Mode.

Not sure if Michael O. is aware that he needs to implement the second to - Tridge may of spoken with him I'm not sure.  Michael O. is on holiday's for the next 2 weeks.

hamishwillee, 2015-10-14:
Thanks @gmorph . I've updated documentation for [MAV_CMD_DO_AUTOTUNE_ENABLE](http://planner.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_do_autotune_enable) to say:

> This command sets the Plane to [AUTOTUNE](http://plane.ardupilot.com/wiki/autotune-mode/) mode.

I already linked it from "Plane Commands in Guided Mode". If it is OK to leave then I will - because it helps me track what has been done/not done (otherwise it looks like this message is an omission).

I'll leave this open and have a look at mission planner in a couple of weeks to see if the commands above went in.

brunoolivieri, 2019-07-15:
Are those parameters auto generated by the wiki builder nowadays?

If so, the necessity remain only in to add documentation. Is that correct?

Hwurzburg, 2020-02-29:
complete review and re-org in progress...dont think this is needed any more...