ArduPilot/ardupilot_wiki #173: TECS (Total Energy Control System) - ARSPD_FBW_MAX :  ARSPD_FBW_MAX.

bluepup, 2015-10-07:
[TECS (Total Energy Control System) for Speed and Height Tuning Guide](http://plane.ardupilot.com/wiki/tecs-total-energy-control-system-for-speed-height-tuning-guide/) - 

Shouldn't ARSPD_FBW_MAX be set to something other than ARSPD_FBW_MAX? Is it TRIM_THROTTLE?
'
ARSPD_FBW_MAX : This is the maximum indicated airspeed (in metres/second) that the speed controller will attempt to control to. This should be set to a speed that the aircraft can achieve in level flight with the throttle set to ARSPD_FBW_MAX.
'

gmorph, 2015-10-08:
That paragraph is trying to say don't set ARSPD_FBW_MAX to something huge that the plane has no chance of achieving.  It should be set to something the plane can achieve in level flight.  I think it should say THR_MAX i.e. "can achieve in level flight with the throttle set to THR_MAX"

priseborough, 2015-10-08:
Yes that is correct.

hamishwillee, 2015-10-08:
@priseborough @gmorph The [ARSPD_FBW_MAX ](http://plane.ardupilot.com/wiki/arduplane-parameters/#maximum_airspeed_arduplanearspd_fbw_max) parameter documentation says:

> This is the maximum airspeed that you want to allow for your airframe in auto-throttle modes. You should ensure that this value is sufficiently above the ARSPD_FBW_MIN value to allow for a sufficient flight envelope to accurately control altitude using airspeed. A value at least 50% above ARSPD_FBW_MIN is recommended.

I think the recommendations here are more useful than the official docs. How about we change the official version to:

  > This is the maximum airspeed that the autopilot will use in auto-throttle modes. It should be set to the highest speed that the aircraft can achieve in level flight with the throttle set to THR_MAX. It must be sufficiently above the ARSPD_FBW_MIN value to allow the autopilot to accurately control altitude using airspeed (at least 50% above ARSPD_FBW_MIN is recommended).

I also think it is "a bad idea" to have duplicate documentation of parameters. So how about we replace [complete parameter list](http://plane.ardupilot.com/wiki/tecs-total-energy-control-system-for-speed-height-tuning-guide/#complete_parameter_list) in the tuning guide with a flat list of links to the official parameters (and perhaps change section name to "Relevant parameter list")?

hamishwillee, 2015-10-13:
@priseborough @gmorph Can you please comment on the above suggestion: https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/173#issuecomment-146711899

priseborough, 2015-10-13:
The suggestion to link to the code comments is a good one. I'm happy to do a pass over the parameter documentation in code to check if it is still up to date.

gmorph, 2015-10-13:
I see what you mean now.  You want to remove the entire "Setting Initial Parameters" bit and instead link to the parameters page.  As it stands I don't want to do that.  The wiki page has a LOT more info then the standard parameters page - just look at THR_MAX for example.  And a lot of advice on how to tune the parameters for this scenario.  I'd rather keep it.
Thanks, Grant.

priseborough, 2015-10-13:
my interpretation was that the parameter descriptors would be referenced, but the other supporting info would remain.

hamishwillee, 2015-10-13:
Grant correctly interpreted my intention. I won't do it now, but will additionally link.

hamishwillee, 2015-10-14:
OK. So I have updated the ARSPD_FBW_MAX comment *in this page* to fix the original bug reported - [see this diff](http://plane.ardupilot.com/wp-admin/revision.php?from=9067&to=9270). 

I have also more broadly updated links and layout ([see this diff](http://plane.ardupilot.com/wp-admin/revision.php?from=9270&to=9272)).

The main problem the changes fix was that the parameters information was duplicated *within the document* (including the "Complete Parameter list". In some cases these were out of sync. I have made the first definition (for example in "Fine tuning") the only definition in this page, linking to the arduplane parameters. The "Complete" parameter list contains just a link to the first definition unless it contains the "only" definition. All the other links in the page go to the first definition of a parameter.

1. One parameter has no definition: TECS_CLMB_MIN - @priseborough @gmorph - can we just copy in the ardupilot parameter version or do you want to provide some words?

Otherwise I consider this issue closed. 

As a separate issue, @priseborough it would still be great if you could:

>  do a pass over the parameter documentation in code to check if it is still up to date.

brunoolivieri, 2019-07-15:
I understand that the page migrated from the WP to the actual Wiki. It also have been updated to:

"ARSPD_FBW_MAX should be set to the maximum speed your aircraft is capable of in level flight with the throttle set to THR_MAX. ARSPD_FBW_MIN should be set to the slowest speed your aircraft can safely fly without stalling in level flight."

Moreover, there are instructions regarding other initial parameters and also the flight testing and tunning.

Would be that enough?

Hwurzburg, 2019-10-10:
I believe the TECS dosc are current and accurate...since there has been no more activity on this for years, I am closing it

closed 2019-10-10 by Hwurzburg