ArduPilot/ardupilot_wiki #175: TRAINING Mode

marcmerlin, 2015-10-12:
[TRAINING Mode](http://plane.ardupilot.com/wiki/training-mode/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

As of 3.4, it also includes stall prevention
https://github.com/diydrones/ardupilot/blob/master/ArduPlane/release-notes.txt

You can also update affected modes in 
http://plane.ardupilot.com/wiki/stall-prevention/

Thanks

hamishwillee, 2015-10-13:
Thanks @marcmerlin!

Both articles were updated along the lines you suggested (I also did some subediting). Please check you are happy:

* [Stall Prevention](http://plane.ardupilot.com/wiki/stall-prevention/) ([diff](http://plane.ardupilot.com/wp-admin/revision.php?from=9062&to=9254)) - added "Training mode" to list of supported modes
* [Training Mode](http://plane.ardupilot.com/wiki/training-mode/) ([diff](http://plane.ardupilot.com/wp-admin/revision.php?from=8788&to=9256)) added information as per diff.

hamishwillee, 2015-10-13:
Closing as is approved by GrantM

closed 2015-10-13 by hamishwillee