ArduPilot/ardupilot_wiki #178: Continuous Pan Servo Support

artemen, 2015-10-13:
Continuous Pan Servos are expensive, however almost ANY regular servo can be EASILY, WITHOUT soldering gun/ programmer modified for continuous rotation use. Please, add such support!
https://www.youtube.com/watch?v=XUm2nj7te8E

hamishwillee, 2015-10-14:
This is primarily a hardware hack - nothing to do with software. There isn't anything particularly relevant to ardupilot here and certainly not something for the wiki.

closed 2015-10-14 by hamishwillee

artemen, 2015-10-14:
Hamish, with all respect you didn't underatand the request: add continuous pan servo support to Antenna Tracker firmware for APM/Pixhawk. 
Please reopen this issue as this is certanly a valid request.

hamishwillee, 2015-10-14:
Hi @artemen 

Thanks - can you please repost this request in [the ArduPilot issue tracker](https://github.com/diydrones/ardupilot/issues) (this issue tracker is for improvements to the wiki infrastructure and content - not to the autopilot itself).

artemen, 2015-10-14:
Done. Thx!