ArduPilot/ardupilot_wiki #179: Windows APM build pages need technical review

hamishwillee, 2015-10-13:
These are the relevant sections:

* [Building ArduPilot for APM2.x on Windows with Arduino](http://dev.ardupilot.com/wiki/building-ardupilot-with-arduino-windows/)
* [Building ArduPilot for APM2.x with Make](http://dev.ardupilot.com/wiki/building_with_make/#windows)
  * [Building ArduPilot for APM2.x on Windows with Make](http://dev.ardupilot.com/wiki/building-ardupilot-for-apm2-x-on-windows-with-make/)

The equivalent "Eclipse" documents should also be reviewed - e.g. [Building ArduPilot for APM2.x with Eclipse on Windows](http://dev.ardupilot.com/wiki/building-apm2-with-eclipse-on-windows/)

gmorph, 2016-02-11:
Looks fine.

closed 2016-02-11 by gmorph

hamishwillee, 2016-02-11:
Thanks Grant!