ArduPilot/ardupilot_wiki #181: UPdate compass docs to make usage more clear

hamishwillee, 2015-10-14:
There are a bunch of compass docs. NOt all of them are very simple or provide clear direction. Review/Update/simplify.

[common-3DR UBlox GPS + Compass Module](http://planner.ardupilot.com/wiki/common-installing-3dr-ublox-gps-compass-module/)

* [PX4 section](http://planner.ardupilot.com/wiki/common-installing-3dr-ublox-gps-compass-module/#connecting_to_px4) it says `Next connect with the Mission Planner and go to INITIAL SETUP | Mandatory Hardware | Compass and set the Orientation to ROTATION_ROLL_180` - does this apply only to PX4?
* GPS configuration information in the [Pixhawk section](http://planner.ardupilot.com/wiki/common-installing-3dr-ublox-gps-compass-module/#connecting_to_pixhawk) : "If the parameters are set-up as shown below, ". This is generic - should it go its own section "Configuring GPS" ?

[common-Advanced Compass Setup](http://planner.ardupilot.com/wiki/common-compass-setup-advanced/)

[Compass Calibration in Mission Planner](http://copter.ardupilot.com/wiki/compass-calibration-in-mission-planner/) (copter)

hamishwillee, 2015-10-14:
Did first update on [common-3DR UBlox GPS + Compass Module](http://planner.ardupilot.com/wiki/common-installing-3dr-ublox-gps-compass-module/)

Still have these questions:

* [PX4 section](http://planner.ardupilot.com/wiki/common-installing-3dr-ublox-gps-compass-module/#connecting_to_px4) it says `Next connect with the Mission Planner and go to INITIAL SETUP | Mandatory Hardware | Compass and set the Orientation to ROTATION_ROLL_180` - does this apply only to PX4?
* GPS configuration information in the [Pixhawk section](http://planner.ardupilot.com/wiki/common-installing-3dr-ublox-gps-compass-module/#connecting_to_pixhawk) : "If the parameters are set-up as shown below, ". This is generic - should it go its own section "Configuring GPS" ?

Hwurzburg, 2019-11-10:
Closing....compass has been extensively re-written since this issue was opened...

closed 2019-11-10 by Hwurzburg