ArduPilot/ardupilot_wiki #182: Communicating with Raspberry Pi via MAVLink

gchiham, 2015-10-14:
[Communicating with Raspberry Pi via MAVLink](http://dev.ardupilot.com/wiki/raspberry-pi-via-mavlink/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

Chambana, 2016-01-31:
Here is an SMS telemetry project for ardupilot/dronekit I've been working on. Should provide limited downstream telemetry (about 1 SMS/second containing 3-4 mavlink messages) and the ability to change modes (e.g. RTL, auto, etc). Can't do waypoint planning over SMS yet, but it's certainly possible. 
 
https://github.com/Chambana/DroneKit-TextMessageTelemetry

hamishwillee, 2016-01-31:
This is very cool, though seems more "DroneKit" than ArduPilot to me. 

@rmackay9 @gmorph do you see any reasonable place we could/should link to this from the wiki?

Chambana, 2016-02-01:
@hamishwillee  Thanks and, yes, it's definitely DroneKit  :)

hamishwillee, 2016-02-01:
Thanks @Chambana , then it probably would have been better in https://github.com/dronekit/dronekit-python/issues :-) Either way, I still need to work out if we can cross link it!

hamishwillee, 2016-02-01:
What I've done with this is [update the readme](https://github.com/dronekit/dronekit-python#users-and-contributors-wanted) with a [link to the forums](https://discuss.dronekit.io/t/notable-projects-using-dronekit/230) where we ask about projects that use dronekit. The first response provides a link to and a summary to your project. 

Thanks very much - we're interested in what people are doing with the tool, and you've made me do something about that!

I'm now closing this issue.

closed 2016-02-01 by hamishwillee