ArduPilot/ardupilot_wiki #183: Plane 3.4 updates

hamishwillee, 2015-10-14:
Plane 3.4 et al have added some new functionality that may need to be documented. Action to go through [release notes](https://github.com/diydrones/ardupilot/blob/master/ArduPlane/release-notes.txt) and check. Also to talk to Tridge.

- [ ] [common-Lightware SF02 Rangefinder](http://plane.ardupilot.com/wiki/common-rangefinder-sf02/) 

  *  Support for both serial and i2c in plane 3.4.0. I2C is not documented. priseborough is investigating - chase up to see if got working

- [ ]  [TRAINING Mode](http://plane.ardupilot.com/wiki/training-mode/) adds support for stall prevention. 
  * Addressed by #175

hamishwillee, 2015-10-14:
@tridge @gmorph If there are any specific issues you know about with the docs for Plane 3.4, please add them here.

auturgy, 2017-10-15:
overtaken by events (aka now at AP3.8.x)

closed 2017-10-15 by auturgy