ArduPilot/ardupilot_wiki #188: add this video to ardurover/boat wiki page?

rmackay9, 2015-10-23:
We don't have many videos of arduboat.. maybe we should add this:

https://youtu.be/Ad7d_S9cHiU
http://diydrones.com/profiles/blogs/arduboat-mk4

hamishwillee, 2015-10-23:
@rmackay9 @gmorph Very cool. 

I've added a [Video Demo](http://rover.ardupilot.com/wiki/rover-video-demonstrations/) page with this - the page is for "what a great project" videos, not "how to use the platform". If either of you can point to other videos of that nature it would be helpful. Essentially anything that shows the platform is more than just a toy.

The structure of the page is such that I expect people to navigate using the page table of contents. Each video has its own section so that, should we choose, we can add words explaining why the video is so cool. I haven't chosen to do so yet.

I also updated the Rover home page slightly to ensure we now talk about both boat and ground vehicles. This links to the video demos. I did try to find a structure that would allow me to add the videos to the home page, but decided that it can't be done without a bit more thought.

So - @gmorph please confirm you're happy with that new page and the changes to the home page. I'll create a separate issue for more extensive changes. Both of you - any other videos you'd like to add?

hamishwillee, 2015-10-26:
I am closing this as done (thanks Randy for videos). Please keep an eye out for more cool stuff like this.

closed 2015-10-26 by hamishwillee

gmorph, 2015-10-26:
Yep I am happy.  Looks great.  Thanks!