ArduPilot/ardupilot_wiki #189: Rover home page updates

hamishwillee, 2015-10-23:
The [Rover home page](http://rover.ardupilot.com/) is a little tired looking, and much less informative than the [Copter home page](http://copter.ardupilot.com/).  I would like to copy the Copter structure with system components, where next, vehicle types etc.

We'd keep the starting sentence because I like that more than what Copter does:
>  "Rover is an advanced open source autopilot for guiding ground vehicles and boats. It can run fully autonomous missions that are defined using mission planning software or pre-recorded by the driver during a manual run."

We would make the top image much smaller file - it is pretty but too slow to load.

@gmorph To do this it would be useful to know more "key selling points" and also "use cases" that Rover meets that we know about (similar to the "Rotor crafts types" section on copter). So we know that it is used for agricultural purposes - any others? Are there any features that make it really stand apart? Should we mention dronekit?

The sparkfun win was so 2014. Is there anything cool we can say about the platform more recently?

Ideas welcome. Is this a good one?

auturgy, 2017-10-15:
@khancyr can you put a PR link in here that we can close against?

khancyr, 2017-10-19:
@auturgy in fact, we speak about that this summer with Randy, and the rover wiki need a complet rework ! Plenty things aren't up to date !

auturgy, 2017-10-24:
ok, no worries - thanks for your efforts with this by the way!

Hwurzburg, 2019-10-10:
guys....there have been a number of updates to Rover recently....anything else need work? otherwise I will close this next week...

closed 2019-11-02 by Hwurzburg