ArduPilot/ardupilot_wiki #190: Building ArduPilot for Pixhawk/PX4 on Windows with Make

molyswu, 2015-10-24:
[Building ArduPilot for Pixhawk/PX4 on Windows with Make](http://dev.ardupilot.com/wiki/building-px4-with-make/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant
Building ArduPilot for Pixhawk/PX4 on Windows with Make,to update firmware for Pixhawk/PX4-v2 board,but  board can't cennect with  missionplaner .why?
Thank for your help!

molyswu, 2015-10-24:
Use missionplaner  update firmware , Pixhawk/PX4-v2 board  no issues.
Thank you for your help!

hamishwillee, 2015-10-25:
For technical support, please post on the [APM forums](http://ardupilot.com/forum/). This issue tracker is where you should report documentation issues/feedback.

closed 2015-10-25 by hamishwillee