ArduPilot/ardupilot_wiki #193: common-PX4FLOW Optical Flow Camera Board Overview

CraigElder, 2015-10-26:
Warning:
The PX4FLOW is not supported in the current stable releases of Copter, Plane or Rover software, but is now supported as an experimental feature in Copter 3.3 beta. It is unlikely that it will ever be supported for use with the APM2.

This section needs to be updated to be consistent with the Copter 3.3 release

hamishwillee, 2015-10-26:
Thanks @CraigElder 

I have updated this to: "The PX4FLOW is supported as an experimental feature starting from Copter 3.3. It is not supported in Plane or Rover."

(This removes the note about APM2 which I consider unnecessary because Copter 3.3 does not support APM2 any more)

@rmackay9  Is this still an "experimental" feature or is it a "proper" feature in Copter 3.3?

hamishwillee, 2015-10-26:
@gmorph Is there any likelihood/plan to put this in Plane? If not, should this really be a common page?

rmackay9, 2015-10-28:
@hamishwillee, .. you know.. I think it's still a bit experimental.

hamishwillee, 2015-10-28:
Thanks @rmackay9 - the current text is OK then. 

@gmorph - is it worth making this "Copter only"? Basically if you're not going to add support in Plane/Rover we might as well.

gmorph, 2015-10-28:
Rover has a long term plan to add support for this but its for indoor navigation purposes.
Plane will probably get it if its not working already so I'd leave it there for plane.
Thanks, Grant.

hamishwillee, 2015-10-28:
Thanks guys. Closing then with current (updated) text.

closed 2015-10-28 by hamishwillee