ArduPilot/ardupilot_wiki #194: Using ROS/Gazebo Simulator with SITL - Installation Instructions Issues/Inconsistencies

sahiljuneja, 2015-10-26:
Hello,

I am trying to install the ROS/Gazebo Simulator. However I noticed some issues with the instructions provided. These are very minor "issues", but can be improved for beginners or people unfamiliar with ROS to an extent as per me. 

Are all the instructions in "Cloning the necessary ROS packages" required to be executed under catkin_ws/src/ ? Because after cloning rotors_simulator.git you need to checkout sonar_plugin, which can be done only when you are in the rotors_simulator folder. This is followed by the wstool instructions. Are they supposed to be executed under /src/ ?

"wstool update -t src rosdep install --from-paths src --ignore-src --rosdistro indigo -y"

What exactly is --from-paths here? As per http://wiki.ros.org/indigo/Installation/Source#Resolving_Dependencies it's not supposed to be replaced by any actual path and is just another option/flag to set. But it throws an error at me: "no such option: --from-paths"

So what step(s) am I missing here?

hamishwillee, 2015-10-26:
Hi @sahiljuneja 

Reading the docs, I am 95% sure that there is a typo in the documets. I think this :
```
wstool update -t src rosdep install --from-paths src --ignore-src --rosdistro indigo -y
```
Should be two lines:
```
wstool update -t src 
rosdep install --from-paths src --ignore-src --rosdistro indigo -y
```
Can you try that and confirm if it works? 

@alexbuyval could you please comment on this bit bit of the question

> Are all the instructions in "Cloning the necessary ROS packages" required to be executed under catkin_ws/src/ ? Because after cloning rotors_simulator.git you need to checkout sonar_plugin, which [...]

sahiljuneja, 2015-10-26:
Hi @hamishwillee  - You're quite right. Those are two separate lines/commands as you point them out. Quite a bit of an overlook from my end too, should have spotted that myself. It's working based on your suggestion now. Thank you. (I assume you can update that wiki?)

hamishwillee, 2015-10-26:
@sahiljuneja Great! Wiki updated now.  

Does that bit of text I asked alexbuyval to comment on still need a response - sounds like maybe not?

sahiljuneja, 2015-10-26:
@hamishwillee - I am trying to confirm by having a simulation up and running. Unfortunately the final step of the installation: running catkin_make, pops up errors - 

> Could not find a package configuration file provided by "mav_msgs" with any of the following names:

> mav_msgsConfig.cmake
> mav_msgs-config.cmake

I am not sure if this is related to the text you asked alexbuyval about or not. I will try to figure this out and update asap.

hamishwillee, 2015-10-27:
@sahiljuneja Thanks for letting me know. Lets see what @alexbuyval says - it is his article and he has had this running before.

alexbuyval, 2015-10-27:
Hi @sahiljuneja and @hamishwillee 

> Are all the instructions in "Cloning the necessary ROS packages" required to be executed under
> catkin_ws/src/ ? Because after cloning rotors_simulator.git you need to checkout sonar_plugin, which [...]

Yes, you need to execute 'git checkout sonar_plugin' under 'catkin_ws/src/rotors_simulator' folder. 

> Could not find a package configuration file provided by "mav_msgs" with any of the following names:

I am sorry, I forgot one repository. 
You need to execute the following command under catkin_ws/src folder:
```
git clone https://github.com/PX4/mav_comm.git
```
Best Regards,
Alex

sahiljuneja, 2015-10-27:
Thanks for the reply @alexbuyval 

Running catkin_make after that is still showing errors. This time -

> Could not find a package configuration file provided by "octomap_msgs" with any of the following names:

> octomap_msgsConfig.cmake
> octomap_msgs-config.cmake

I can go ahead and install these packages myself. But I have two questions -

For any such errors should I stick with the source provided on the official ROS Wiki page to download the package? (http://wiki.ros.org/octomap_msgs) Because for mav_comm the source mentioned in ROS wiki was different from your suggestion.

Do you think all these packages need to be mentioned in the dev page - http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/
Because I think they should be mentioned for someone who is trying to use ROS just for this. I guess a lot of students or researchers would find this helpful in the long run and it's better to add what packages might be required that weren't installed with ROS itself. Your feedback on this @hamishwillee  is also welcome!

hamishwillee, 2015-10-27:
1. Can you confirm the changes below match what you actually did (marked in red)

   [image]

2.
   > Do you think all these packages need to be mentioned in the dev page

   Yes. The instructions need to be such that as long as someone follows them "to the letter" the process will work.

sahiljuneja, 2015-10-27:
Yes, I did clone "mav_comm".

And then I did clone "rotors_simulator" in ~/catkin_ws/src/

Since you are cloning it, I don't think you need to separately create a directory first, right? I didn't do that. I just cloned it. Then entered the cloned directory (~/catkin_ws/src/rotors_simulator) and did a checkout on sonar_plugin.

After that I went back to ~/catkin_ws/src/ and executed the wstool related instructions till the end. 

I then ran a catkin_make in ~/catkin_ws/ but that pooped up a new error as per my previous comment.

hamishwillee, 2015-10-27:
@sahiljuneja Thanks! I updated [the instructions](http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/#cloning_the_necessary_rospackages) in line with your comment above. I know there are still problems - hopefully @alexbuyval can help (I'm updating the doc as we go along to make sure I don't miss anything).

sahiljuneja, 2015-10-28:
Hey @hamishwillee and @alexbuyval 

I am making a list of missing packages I had to install after following the instructions provided on the [before this thread's] dev page.

1. git clone https://github.com/PX4/mav_comm.git

2. git clone https://github.com/OctoMap/octomap_msgs.git

3. git clone https://github.com/ethz-asl/glog_catkin.git

4. git clone https://github.com/catkin/catkin_simple.git

After that I ran catkin_make command in ~/catkin_ws/

It builds. There were some errors which were due to some backward compatibility. I can't replicate that error at the moment (scroll limit). I will try to build it again and then paste the error later.

I then try to follow the command to run the simulator.

> sim_vehicle.sh -f arducopter_sitl_ros --console

This pops up another error - 

> sim_vehicle.sh: command not found

(I even tried with sudo just in case)

So I guess there's something else that's missing?

This is the farthest I have reached after installing the above packages on my own. I will update if I can figure out sim_vehicle.sh issue or if you could let me know why this would happen.

alexbuyval, 2015-10-28:
Hi @sahiljuneja and @hamishwillee 

Regarding list of missing packages. I have just checked. The 'glog_catkin' and 'catkin_simple' are not installed in my OS. The 'octomap_msgs' is installed, although I don't understand where it is used. Anyway, I think, it is better if you install 'octomap_msgs' from ROS repository (instead of cloning from source) by the following command:
```
sudo apt-get install ros-indigo-octomap-msgs
```
> sim_vehicle.sh: command not found

Try to add the following line into .bashrc file:
```
export PATH=$PATH:$HOME/diydrones/ardupilot/Tools/autotest
```
Please note, you should use your path to ardupilot folder.

sahiljuneja, 2015-10-28:
That's interesting that you don't require glog_catkin and catkin_simple. Those popped up when trying to run a catkin_make, so I am not sure why there would be a difference here. 

And thank you for the PATH related help. The sim_vehicle.sh related command now runs and builds everything before trying to run the Simulation. But there were errors.

> ERROR: cannot launch node of type [mavros/mavros_node]: can't locate node [mavros_node] in package [mavros]
> ERROR: cannot launch node of type [joy/joy_node]: joy

Those are the errors in RED when the xterm console opens up upon running the sim_vehicle.sh related command.

It also states somewhere there that 

> ImportError: No module named mav_msgs.msg

and also

> ImportError: No module named mavros.msg

I thought mav_comm should have handled that. But it looks like it's not importing mav_msgs.msg module from the mav_comm package. (~/catkin_ws/src is in my ROS_PACKAGE_PATH at the moment). So is this another path related issue?

sahiljuneja, 2015-10-28:
Okay so I figured out the issue with the above.

I forgot to run

> source ~/catkin_ws/devel/setup.bash
after running catkin_make

I ran the sim_vehicle.sh command, and this time there were some errors pertaining to Gazebo I believe -

> Error [Node.cc:90] No namespace found

The process dies after throwing an instance of 'Ogre::InvalidParametersException'. Gazebo did open up, but nothing spawned there, and it basically crashed.

It's unclear if this problem is from my end because of Gazebo that I have installed, or it's because of the code pertaining to the Simulation. If you could point that out, that would be really helpful.

hamishwillee, 2015-10-29:
So I have added octomap by renaming "Cloning" section to include "installation" and adding line as shown:
[image]

Following that section, I've added information on setting up the path to the build tools:
[image]

OK?

alexbuyval, 2015-10-29:
Hi @sahiljuneja and @hamishwillee 

First of all, I have cleaned my all ROS packages and rebuild all again. I have found that 'glog_catkin' and 'catkin_simple' packages are necessary for 'rotor_simulator' package. So we need to add:
```
git clone https://github.com/ethz-asl/glog_catkin.git
git clone https://github.com/catkin/catkin_simple.git
```

@hamishwillee 
>OK?

Yes, it is.

@sahiljuneja 
>However the simulation process still closes on me
>ImportError: No module named mav_msgs.msg

I think that new terminal window with simulation process which is run automaticly, know nothing about ROS environment. So you need according ROS install instruction to execute the following command:
```
echo "source /opt/ros/indigo/setup.bash" >> ~/.bashrc
```
and also to execute the the following command:
```
echo "source /home/youruser/catkin_ws/devel/setup.bash" >> ~/.bashrc
```
In this case, you don't need to run each time.
```
source ~/catkin_ws/devel/setup.bash
```

sahiljuneja, 2015-10-29:
@alexbuyval - I already had those in my .bashrc and it didn't change anything,

But I did figure out the issue and why I was getting those errors. It's the wstool commands to have mavros downloaded and installed. In the instructions page it's not specified where to run those commands, so I ran them in ~/catkin_ws/src 

That led to mavros being ending up in 

> ~/catkin_ws/src/src/mavros

Yes, two src folders. Hence the dependency and import issues. 

**So this is what I did.**

I removed the other src folder. And ran the wstool commands while in ~/catkin_ws

Followed by running the final rosdep install command as given. However this threw an error regarding glog_catkin. So instead I simply went inside ~/catkin_ws/src and ran 

> rosdep install mavros

After that (while in ~/catkin_ws/src), I ran

> cd ..
> catkin_make
> source devel/setup.bash

and then ran the sim_vehicle.sh command in the ArduCopter folder.

And ***viola***! The simulation ran. Well somewhat. The model of the quadcopter generated has propellers which are ***not*** aligned with the motors. And that's just one of the issues. I got new errors, which I can't fix at the moment, and they are - 

> Error [SystemPaths.cc:367] File or path does not exist["/home/sahil/catkin_ws/src/rotors_simulator/rotors_description/meshes/sonar_sensor/max_sonar_ez4.dae"]
Error [Visual.cc:2133] No mesh specified
Warning [Visual.cc:819] Unable to get Material[artag] for Geometry[ARtag1::ARtag::ARtag_visual. Object will appear white
Warning [Visual.cc:819] Unable to get Material[artag] for Geometry[ARtag1::ARtag::ARtag_visual. Object will appear white
Error [SystemPaths.cc:367] File or path does not exist["/home/sahil/catkin_ws/src/rotors_simulator/rotors_description/meshes/sonar_sensor/max_sonar_ez4.dae"]
Error [Visual.cc:2133] No mesh specified
[ INFO] [1446160217.783057571, 15.605000000]: CON: Got HEARTBEAT, connected.
[ERROR] [1446160217.783629596, 15.605000000]: FCU: Initialising APM...
[ INFO] [1446160218.973956634, 16.790000000]: VER: 1.1: Capabilities 0x0000000000000000
[ INFO] [1446160218.974094490, 16.790000000]: VER: 1.1: Flight software:     03040000 (21f00ddf)
[ INFO] [1446160218.974143155, 16.790000000]: VER: 1.1: Middleware software: 00000000 (        )
[ INFO] [1446160218.974183781, 16.790000000]: VER: 1.1: OS software:         00000000 (        )
[ INFO] [1446160218.974225282, 16.790000000]: VER: 1.1: Board hardware:      00000000
[ INFO] [1446160218.974265925, 16.790000000]: VER: 1.1: VID/PID: 0000:0000
[ INFO] [1446160218.974305523, 16.790000000]: VER: 1.1: UID: 0000000000000000
[ WARN] [1446160218.974483308, 16.790000000]: CMD: Unexpected command 520, result 0
[ERROR] [1446160220.854330033, 18.655000000]: FCU: Calibrating barometer
[ERROR] [1446160222.940636729, 20.720000000]: FCU: Initialising APM...
[ERROR] [1446160228.194757530, 25.925000000]: FCU: Initialising APM...
[ INFO] [1446160232.913031931, 30.620000000]: WP: mission received
[ERROR] [1446160233.234321611, 30.930000000]: FCU: Initialising APM...
[ERROR] [1446160238.266410108, 35.935000000]: FCU: Initialising APM...
[ERROR] [1446160243.369001037, 41.000000000]: FCU: Initialising APM...
[ERROR] [1446160246.951326002, 44.545000000]: FCU: barometer calibration complete
[ WARN] [1446160249.021900887, 46.605000000]: PR: request list timeout, retries left 2
[ WARN] [1446160250.031821424, 47.605000000]: PR: request list timeout, retries left 1
[ WARN] [1446160251.039997389, 48.605000000]: PR: request list timeout, retries left 0
[ WARN] [1446160256.715002445, 54.240000000]: CON: Lost connection, HEARTBEAT timed out.
[ERROR] [1446160268.935108965, 66.375000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[ERROR] [1446160268.935270215, 66.375000000]: FCU: Frame: QUAD
[ERROR] [1446160268.935385712, 66.375000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[ERROR] [1446160268.935526337, 66.375000000]: FCU: Frame: QUAD
[ERROR] [1446160268.935642155, 66.375000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[ERROR] [1446160268.935736730, 66.375000000]: FCU: Frame: QUAD
[ERROR] [1446160268.935836412, 66.375000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[ERROR] [1446160268.935918810, 66.375000000]: FCU: Frame: QUAD
[ INFO] [1446160272.627918939, 70.050000000]: CON: Got HEARTBEAT, connected.
[ INFO] [1446160273.656397769, 71.065000000]: VER: 1.1: Capabilities 0x0000000000000b83
[ INFO] [1446160273.656482695, 71.065000000]: VER: 1.1: Flight software:     03040000 (21f00ddf)
[ INFO] [1446160273.656523966, 71.065000000]: VER: 1.1: Middleware software: 00000000 (        )
[ INFO] [1446160273.656561554, 71.065000000]: VER: 1.1: OS software:         00000000 (        )
[ INFO] [1446160273.656600935, 71.065000000]: VER: 1.1: Board hardware:      00000000
[ INFO] [1446160273.656633877, 71.065000000]: VER: 1.1: VID/PID: 0000:0000
[ INFO] [1446160273.656669873, 71.065000000]: VER: 1.1: UID: 0000000000000000
[ WARN] [1446160273.656809234, 71.065000000]: CMD: Unexpected command 520, result 0
[ INFO] [1446160287.844241635, 85.165000000]: WP: mission received
[ERROR] [1446160303.021452816, 100.240000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[ERROR] [1446160303.021602580, 100.240000000]: FCU: Frame: QUAD
[ WARN] [1446160320.698504590, 117.780000000]: FCU: PreArm: RC not calibrated
[ERROR] [1446160333.132914325, 130.155000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[ERROR] [1446160333.133110153, 130.155000000]: FCU: Frame: QUAD
[ERROR] [1446160363.348979525, 160.115000000]: FCU: APM:Copter V3.4-dev (21f00ddf)
[... cut ...]
[ INFO] [1446160481.238236875, 277.155000000]: PR: parameters list received
[ WARN] [1446160482.320416257, 278.215000000]: FCU: PreArm: RC not calibrated

@hamishwillee - So from the instructions viewpoint, it was the package mavros not being in the correct folder. For that I believe the wstool and rosdep commands need to be modified. I think Alex can give a better idea on how to modify them. And these new errors, since they are in reference to the simulation itself, should probably be raised as a separate issue (if I can't fix them on my own)?

hamishwillee, 2015-10-30:
@sahiljuneja @alexbuyval I added the missing clone commands for glog_catkin.git and catkin_simple.git

Sounds like the other changes might require that you have actually "done the job". @alexbuyval - can you update them appropriately in line with Sahil's comments?

alexbuyval, 2015-10-31:
@hamishwillee I have just updated one command in clone/install section. I have replaced 'cd ..' on 'cd ../..'

@sahiljuneja  The properlers are not aligned with the motors, because I have used physical parameters for 450 mm frame, but for graphics I have used the Iris drone mesh. So 3D graphics does not match simulated model. It is not critical. 

> Error [SystemPaths.cc:367] File or path does not exist["/home/sahil/catkin_ws/src/rotors_simulator/rotors_description/meshes/sonar_sensor/max_sonar_ez4.dae"]

It is not critical too. However, I have fixed the path into 'sonar_sensor.urdf.xacro'. Just update 'rotors_simulator' repository.

>Warning [Visual.cc:819] Unable to get Material[artag] for Geometry[ARtag1::ARtag::ARtag_visual. Object will appear white

This is critical for control node. The visual marker will appear white.
It seems that 'catkin_ws/src/rotors_simulator/rotors_description/Media' path is not added into the Gazebo media path. I have added some dependencies into package.xml. It should help. Just update 'rotors_simulator' repository.

hamishwillee, 2015-11-01:
Hi @alexbuyval 

Thank for that - I published the change. @sahiljuneja can you please confirm that this is working for you now?

Cheers
H

sahiljuneja, 2015-11-02:
Hi @hamishwillee - Yes it seems everything is working in regards to the simulation instructions.

Thank you for all your help on this. You too @alexbuyval 

I did wish to ask something else.

How can I stay updated with the development going on with ROS/Gazebo based ArduCopter Simulation? Or is this the extent of what is planned?

hamishwillee, 2015-11-03:
Hi @sahiljuneja @alexbuyval 

1. Fantastic. Thank you both for your patience and persistence on this

> How can I stay updated with the development going on with ROS/Gazebo based ArduCopter Simulation? Or is this the extent of what is planned?

@alexbuyval is probably the best person to ask that question. As far as I know all of the work is community based - so perhaps happened organically rather than planned?

In any case, I am now closing this issue!

closed 2015-11-03 by hamishwillee

alexbuyval, 2015-11-03:
@sahiljuneja This SITL mode is just instrument to solve other tasks. I use it to test a firmware before a real flight. For instance, now I am trying to fuse extrenal messages about visual position into the [EKF](https://groups.google.com/forum/#!topic/drones-discuss/HEYEDZEMUY0).

sahiljuneja, 2015-11-04:
@alexbuyval  When you say instrument to solve other tasks, do you test the firmware as HITL? Or do you do actual implementation of algorithms and run them on the drone in the simulation? That would be awesome to check out!

alexbuyval, 2015-11-05:
@sahiljuneja I meant that I use this SITL mode to check new algorithms into firmware. Of course, I run them on the drone in the simulation.

koustubh-dwivedy, 2016-06-14:
@alexbuyval just a small problem, each time I go through the instructions given on http://ardupilot.org/dev/docs/using-rosgazebo-simulator-with-sitl.html and reach the step `rosdep install --from-paths src --ignore-src --rosdistro indigo -y
`
I get the following error->
```
koustubh@koustubh:~/Desktop/catkin_SITL_ws$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
glog_catkin: Cannot locate rosdep definition for [gflags_catkin]

```
Can you tell me how can I resolve it?
Thanks