ArduPilot/ardupilot_wiki #195: Setting up SITL on Windows 10, can't make SITL

dumle29, 2015-10-27:
I was following the getting started here: http://python.dronekit.io/guide/getting_started.html

I get to
    make sitl -j4

But I can't build it, here is the errors I'm getting:
http://pastebin.com/6yeS7Y23

I have followed the instructions. Here's the specs:

#### Hardware
* Medion Akoya P6638 (MD99170)
* Core i3-3120M
* 8GiB RAM

#### Software
Software | Version
------------|-----------
Windows 10 | Home x64
Cygwin | 2.873
GCC-G++ | 4.9.3-1
GAWK | 4.1.1-1
ccache | 3.1.9-2
autoconf | 13-1
automake | 9-1
git | 2.5.3-1
libtool | 2.4.6-2
make | 4.1-1
libexpat | 2.1.0-3
procps | 3.2.8-4

dumle29, 2015-10-27:
Friend on W7 has the same issues

hamishwillee, 2015-10-27:
@dumle29 That link is to dronekit, not the ardupilot wikis. Are you attempting to follow the instructions in [Setting up SITL on Windows](http://dev.ardupilot.com/wiki/sitl-native-on-windows/) ?

Are you attempting this purely to test dronekit or do you need to be able to do regular builds? If the former there are other ways to test dronekit which I might be able to suggest.

dumle29, 2015-10-27:
Hello, I am following "Setting up SITL on Windows" yes :) I've installed cygwin, and all the packages to my F drive, as my C drive is a small SSD.

The file doesn't exist in /usr/lib/gcc/, however the files are in /usr/lib/gcc/i686-pc-cygwin/4.9.3/include

hamishwillee, 2015-10-27:
Hi @dumle29 

My cygwin packages same except for "gawk 4.1.3-1" - though I very much doubt this is the cause. I just synced to the head revision of ardupilot and rebuilt without problem.

The most likely cause is "bad paths". Did you do this step?
3. Add the following line to the end of .bashrc

   ```
export PATH=$PATH:$HOME/ardupilot/Tools/autotest
```

Also, please confirm you are running the build from within /ardupilot/ArduCopter?

Can you please show me the output of printenv (in cygwin prompt). I am most interested in the paths and home variables. For example, my env looks like:
```
$ printenv
USERDOMAIN_ROAMINGPROFILE=DESKTOP-D5A9KSQ
HOMEPATH=\Users\hamis
APPDATA=C:\Users\hamis\AppData\Roaming
ProgramW6432=C:\Program Files
HOSTNAME=DESKTOP-D5A9KSQ
SHELL=/bin/bash
TERM=xterm
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
PROFILEREAD=true
WINDIR=C:\WINDOWS
PUBLIC=C:\Users\Public
OLDPWD=/home/hamis/ardupilot
ORIGINAL_PATH=/cygdrive/c/Python27:/cygdrive/c/Python27/Scripts:/cygdrive/c/Prog                                 ramData/Oracle/Java/javapath:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/c                                 ygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v                                 1.0:/cygdrive/c/Program Files (x86)/Calibre2:/cygdrive/c/HashiCorp/Vagrant/bin:/                                 cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/Git/mingw64/bin:/cygd                                 rive/c/Program Files/Git/usr/bin:/cygdrive/c/Program Files (x86)/MHV AVR Tools/b                                 in:/cygdrive/c/Program Files (x86)/Skype/Phone:/cygdrive/c/Program Files (x86)/Q                                 uickTime/QTSystem:/cygdrive/c/Users/hamis/AppData/Roaming/Python/Scripts:/cygdri                                 ve/c/Python27:/cygdrive/c/Users/hamis/AppData/Local/Pandoc
USERDOMAIN=DESKTOP-D5A9KSQ
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
OS=Windows_NT
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
ALLUSERSPROFILE=C:\ProgramData
USER=hamis
!::=::\
TEMP=/tmp
COMMONPROGRAMFILES=C:\Program Files (x86)\Common Files
USERNAME=hamis
PROCESSOR_LEVEL=6
ProgramFiles(x86)=C:\Program Files (x86)
PATH=/usr/local/bin:/usr/bin:/cygdrive/c/Python27:/cygdrive/c/Python27/Scripts:/                                 cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/WINDOWS/system32:/cygdri                                 ve/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/Wind                                 owsPowerShell/v1.0:/cygdrive/c/Program Files (x86)/Calibre2:/cygdrive/c/HashiCor                                 p/Vagrant/bin:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/Git/mi                                 ngw64/bin:/cygdrive/c/Program Files/Git/usr/bin:/cygdrive/c/Program Files (x86)/                                 MHV AVR Tools/bin:/cygdrive/c/Program Files (x86)/Skype/Phone:/cygdrive/c/Progra                                 m Files (x86)/QuickTime/QTSystem:/cygdrive/c/Users/hamis/AppData/Roaming/Python/                                 Scripts:/cygdrive/c/Python27:/cygdrive/c/Users/hamis/AppData/Local/Pandoc:/home/                                 hamis/ardupilot/Tools/autotest
PSModulePath=C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules\
PWD=/home/hamis/ardupilot/ArduCopter
SYSTEMDRIVE=C:
PROCESSOR_ARCHITEW6432=AMD64
LANG=en_US.UTF-8
USERPROFILE=C:\Users\hamis
TZ=Australia/Sydney
PS1=\[\e]0;\w\a\]\n\[\e[32m\]\u@\h \[\e[33m\]\w\[\e[0m\]\n\$
LOGONSERVER=\\MicrosoftAccount
CommonProgramW6432=C:\Program Files\Common Files
PROCESSOR_ARCHITECTURE=x86
LOCALAPPDATA=C:\Users\hamis\AppData\Local
ProgramData=C:\ProgramData
EXECIGNORE=*.dll
SHLVL=1
HOME=/home/hamis
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.py;.pyw
OPENSSL_CONF=C:\OpenSSL-Win32\bin\openssl.cfg
HOMEDRIVE=C:
VBOX_MSI_INSTALL_PATH=C:\Program Files\Oracle\VirtualBox\
COMSPEC=C:\WINDOWS\system32\cmd.exe
TMP=/tmp
SYSTEMROOT=C:\WINDOWS
PRINTER=Brother HL-2270DW series
PROCESSOR_REVISION=3a09
FPS_BROWSER_USER_PROFILE_STRING=Default
INFOPATH=/usr/local/info:/usr/share/info:/usr/info
PROGRAMFILES=C:\Program Files (x86)
NUMBER_OF_PROCESSORS=4
asl.log=Destination=file
SESSIONNAME=Console
COMPUTERNAME=DESKTOP-D5A9KSQ
_=/usr/bin/printenv
```

hamishwillee, 2015-10-27:
PS Another way to run SITL natively is to use [dronekit-sitl](https://github.com/dronekit/dronekit-sitl) - see readme on that page for usage. This is still experimental but is very fast and easy to test - good if you don't want to test on development branches!!! 

Basically the main limitation right now of that tool is that it doesn't allow tcp connections - you instead need to connect via tcp. In the examples for example you'd do
```
vehicle_state.py --connect tcp:youripaddress:5760
```

dumle29, 2015-10-28:
Here's the printenv :)

```USERDOMAIN_ROAMINGPROFILE=DESKTOP-LVAKMBM
HOMEPATH=\Users\mikke
APPDATA=C:\Users\mikke\AppData\Roaming
ProgramW6432=C:\Program Files
HOSTNAME=DESKTOP-LVAKMBM
SHELL=/bin/bash
TERM=xterm
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
PROFILEREAD=true
WINDIR=C:\Windows
PUBLIC=C:\Users\Public
OLDPWD=/cygdrive/c/Windows/system32
ORIGINAL_PATH=/cygdrive/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/f/Program Files (x86)/nodejs:/cygdrive/c/Program Files/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit:/cygdrive/c/Program Files/Intel/WiFi/bin:/cygdrive/c/Program Files/Common Files/Intel/WirelessCommon:/cygdrive/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin:/cygdrive/f/Program Files (x86)/MiKTeX 2.9/miktex/bin:/cygdrive/f/Program Files/MATLAB/R2015b/runtime/win64:/cygdrive/f/Program Files/MATLAB/R2015b/bin:/cygdrive/c/Users/mikke/AppData/Local/atom/bin:/cygdrive/c/Users/mikke/AppData/Roaming/npm:/cygdrive/c/Program Files/Intel/WiFi/bin:/cygdrive/c/Program Files/Common Files/Intel/WirelessCommon:/cygdrive/c/Python27:/cygdrive/c/Python27/Scripts:/cygdrive/f/Program Files/Mercurial
USERDOMAIN=DESKTOP-LVAKMBM
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
OS=Windows_NT
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
ALLUSERSPROFILE=C:\ProgramData
USER=mikke
!::=::\
TEMP=/tmp
COMMONPROGRAMFILES=C:\Program Files (x86)\Common Files
VXIPNPPATH=C:\Program Files (x86)\IVI Foundation\VISA\
USERNAME=mikke
PROCESSOR_LEVEL=6
ProgramFiles(x86)=C:\Program Files (x86)
PATH=/usr/local/bin:/usr/bin:/cygdrive/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/f/Program Files (x86)/nodejs:/cygdrive/c/Program Files/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit:/cygdrive/c/Program Files/Intel/WiFi/bin:/cygdrive/c/Program Files/Common Files/Intel/WirelessCommon:/cygdrive/c/Program Files (x86)/IVI Foundation/VISA/WinNT/Bin:/cygdrive/f/Program Files (x86)/MiKTeX 2.9/miktex/bin:/cygdrive/f/Program Files/MATLAB/R2015b/runtime/win64:/cygdrive/f/Program Files/MATLAB/R2015b/bin:/cygdrive/c/Users/mikke/AppData/Local/atom/bin:/cygdrive/c/Users/mikke/AppData/Roaming/npm:/cygdrive/c/Program Files/Intel/WiFi/bin:/cygdrive/c/Program Files/Common Files/Intel/WirelessCommon:/cygdrive/c/Python27:/cygdrive/c/Python27/Scripts:/cygdrive/f/Program Files/Mercurial:/home/mikke/ardupilot/Tools/autotest
PSModulePath=C:\Windows\system32\WindowsPowerShell\v1.0\Modules\
PWD=/home/mikke
SYSTEMDRIVE=C:
PROCESSOR_ARCHITEW6432=AMD64
LANG=en_GB.UTF-8
USERPROFILE=C:\Users\mikke
TZ=Europe/Paris
PS1=\[\e]0;\w\a\]\n\[\e[32m\]\u@\h \[\e[33m\]\w\[\e[0m\]\n\$
LOGONSERVER=\\MicrosoftAccount
CommonProgramW6432=C:\Program Files\Common Files
PROCESSOR_ARCHITECTURE=x86
LOCALAPPDATA=C:\Users\mikke\AppData\Local
ProgramData=C:\ProgramData
EXECIGNORE=*.dll
SHLVL=1
HOME=/home/mikke
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY
HOMEDRIVE=C:
VBOX_MSI_INSTALL_PATH=F:\Program Files\Oracle\VirtualBox\
COMSPEC=C:\Windows\system32\cmd.exe
TMP=/tmp
SYSTEMROOT=C:\Windows
PRINTER=\\mfc-print03.aau.dk\Follow-You
PROCESSOR_REVISION=3a09
OMP_NUM_THREADS=4
FPS_BROWSER_USER_PROFILE_STRING=Default
INFOPATH=/usr/local/info:/usr/share/info:/usr/info
PROGRAMFILES=C:\Program Files (x86)
VS140COMNTOOLS=F:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools\
NUMBER_OF_PROCESSORS=4
SESSIONNAME=Console
COMPUTERNAME=DESKTOP-LVAKMBM
_=/usr/bin/printenv
```

The path seems to be right. 

I'll take a look at dronekit-sitl if I can't get this to run :)

hamishwillee, 2015-10-29:
Hi Mikke, 

I can't see anything wrong with that spec. If you did indeed follow the instructions, I can only assume that Cygwin hates you :-0. 

Further debugging of this is beyond me, because I can't reproduce the issue  (instructions work for me). All is not lost. You can also raise this question on the APM forums (where more people might have run into the problem). 

dronekit-sitl is the fastest and easiest way to get one of the builds up and running. However it is experimental and I don't trust it as much as the normal builds. 

Another alternative is to use [these instructions and do the build in Vagrant](http://dev.ardupilot.com/wiki/setting-up-sitl-using-vagrant/). This method is good if you really do want to **build** the latest sources etc, but you can't get native to work (or if you just want to keep a clean Windows environment). The only limitation is that it doesn't launch the MAVproxy console etc (the VM image is UI-less). That has never felt like a limitation to me!

I am getting a new computer early next week so I can attempt to verify again "from clean" then. Will keep this clean until at least then.

Regards
Hamish

dumle29, 2015-10-29:
Well thanks for the look :) I'll try when I get home again, and try it on my windows partition of my desktop. So far I've only tried my laptop, as it is for a uni project.

Maybe we should look at an anti-hate module to be made available for cygwin :P

hamishwillee, 2015-10-29:
> Maybe we should look at an anti-hate module to be made available for cygwin :P

Wouldn't that be nice! I updated to Windows 10 recently (as an upgrade keeping my settings and it stuffed up Cygwin paths bigtime. Had to reinstall Windows and Cygwin completely to get it to work. Upshot, not everything is working perfectly with these two yet.

hamishwillee, 2015-11-06:
Hi @dumle29

I just retested on my clean machine and everything worked fine. I believe the instructions are good, but that it is possible for Cygwin and Windows to get confused. I don't think that is something that can really be sorted out by the documentation - and in any case if I can't reproduce I can't debug. 

Therefore I am closing "can't reproduce". 

You might be best off asking this question on the forums in case others have run into similar Cygwin problems. Sorry I can't be of more help.

Regards
Hamish

closed 2015-11-06 by hamishwillee

dumle29, 2015-11-11:
I'm going to use linux for that then :) Thanks a lot for taking the time though :)

hamishwillee, 2015-11-13:
@dumle29 You're welcome - sorry I couldn't help you.

hamishwillee, 2015-11-25:
@dumle29 Just FYI, someone had the same problem in @229. In their case it turned out they were not actually downloading all the packages in cygwin correctly.

dumle29, 2016-03-29:
Thanks for the heads up. I think I remember reading up on that issue, but not finding it helped. However I did figure out the issue today. There was a version mismatch between gcc and g++, which is apparently not allowed. both have to be the same version.