ArduPilot/ardupilot_wiki #196: Adding the step on the page of "Adding a New Flight Mode"

zxc2694, 2015-10-27:
We need to add the step, because the console appears the error message after i finished all steps and compiling.
The step is that add two lines "XXX_init(bool ignore_checks);" and "void XXX_run();" in the Copter.h 
(Maybe XXX is "newflightmode").

hamishwillee, 2015-10-28:
Thanks @zxc2694 !

@rmackay9 @gmorph In the instructions [Adding a New Flight Mode (Copter Code Overview)](http://dev.ardupilot.com/wiki/apmcopter-adding-a-new-flight-mode/) (Step2) we define some  _init and _run methods, but do not declare them. All the other modes do the same thing but declare the methods in copter.h. 

Can you confirm that a step to add these methods for a new mode is needed?

zxc2694, 2015-10-28:
Hi @hamishwillee , 
I have confirmed. The project which is added new mode can be compiled successfully when i declared the methods in copter.h.

hamishwillee, 2015-10-29:
I've updated [Adding a New Flight Mode (Copter Code Overview)](http://dev.ardupilot.com/wiki/apmcopter-adding-a-new-flight-mode/) with (new) step 3 as shown below.
[image]

Closing this issue now.

Thanks @zxc2694 !

closed 2015-10-29 by hamishwillee