ArduPilot/ardupilot_wiki #197: Seagull #MAP addition to ArduPilot Wiki - full page

SeagullUAV, 2015-10-30:
Hi,

I am the owner of Seagull UAV, and would like to add a full page with my latest product...

Seagull #MAP is the cheapest solution on the market right now, that offers "Camera On/Off" for Sony MULTI cameras..!

Seagull #MAP - http://www.seagulluav.com/product/seagull-map/
Link to our Support page - http://www.seagulluav.com/seagull-map-support/
Link to Seagull #MAP Manual - http://www.seagulluav.com/wp-content/uploads/2015/08/Seagull-MAP-Manual.pdf

Currently we are placed on this page together with SkySight MONO...
http://copter.ardupilot.com/wiki/common-camera-shutter-triggering-cables-for-purchase/

I am also the previous co-founder of Farsight, and project owner of SkySight MONO.. I have since my time in Farsight started Seagull UAV and released an improved and cheaper solution for Sony MULTI cameras + support for a lot of other brands and models.. :)

Here is a link to some of the info we would like on the "one-pager" - http://www.seagulluav.com/wp-content/uploads/2015/10/Seagull-MAP-ardupilot.docx

Looking forward to your reply !

Best regards,

Rone Scholz-Nielsen
CEO // Founder
Seagull UAV

hamishwillee, 2015-11-01:
Hi Rone (@SeagullUAV )

Currently I tend to think having a full page makes a lot of sense if there isn't decent documentation for how a solution works - for example a cable hack. I'm trying to avoid creating the same for  commercial products because a)you already have a great manual b) having duplicate documentation is likely to lead to the case where one version gets out of date ... and that could well be the wiki one.

If you really want this additional page I can create this, but I would really appreciate your ongoing commitment to making sure that the page remains up to date with your manual (I won't be able to track changes unless you tell me). I'd rather not unless you feel strongly that this is needed.

Let me know what you think!

SeagullUAV, 2015-11-01:
Hi Hamish,

 

First of - Thanks for Your reply ! :)

 

I know it seems extreme to add a full page, but it all comes down to a new business that want to offer the best solution – including the best prices worldwide!

 

I know from a fact that your site gets a lot of traffic from newcomers that seeks some proper guidance on their new project idea.

And hence I want to offer the best product at the best quality to the markets best price, this will help a lot of them to a faster start.

 

Back when I launched SkySight MONO in my previous company, I had almost 60% of all traffic coming from your Wiki! That is definitely something I would like to add to my new startup..

 

We want to focus on modules and technology based around the ArduPilot platform, and Seagull #MAP was the perfect "intro" for us for our "Get to market" strategy..

 

We have a lot of satisfied customers already, but we need the "boost" from your site to help us "get the time" to develop and push out new products!

 

I'm running Seagull UAV as my secondary job with a good friend of mine, and we would really love to take it fulltime at some point – so we spend night and day of our spare time to grow and develop new ideas..

 

I know it will be extra work – first off for You, and later for me should anything change in the product, but I'm willing to take this task on my shoulders!

 

I'm looking forward to your reply :)

 

Have a Greay day !

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

hamishwillee, 2015-11-02:
Hi Rone

I understand this is good advertising :-) I have no problem with supporting this, and it is useful for users not to have to navigate off-site. So unless it becomes burdensome in terms of maintenance we can do this. I'll put this on my list for the next few days.

Thanks
H

SeagullUAV, 2015-11-02:
Hi Hamish,

 

Thank You very much – it's highly appreciated !

 

Let me know if you need my review before posting :)

 

 

Have a Great day !

 

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

hamishwillee, 2015-11-03:
Hi Rone

The page is [live here](http://planner.ardupilot.com/wiki/common-camera-shutter-triggering-for-sony-multiport-connectors-using-seagull-map/) (and on equivalent urls in copter/rover/plane). It follows the patterns of similar pages in the section (not quite what you have because we have some general information pages which we link to).

I think this is good, but please have a quick sanity check.

Regards
Hamish

SeagullUAV, 2015-11-03:
Hi Hamish,

 

It looks AWSOME ! ;) - Thanks for the great work..!

 

I found some minor things that I would like to have changed, but let me do a propper review and get back to you a bit later today..

Thanks again Hamish – I really appreciate it..! :)

 

Have a Great day ! 

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

SeagullUAV, 2015-11-03:
Hi Hamish,

 

I have now made my review, and have the following changes – nothing really drastic :)

 

I've also attached 2 pictures – the screen dumps below tell where they need to be placed/replaced..

 

 

I hope it's easy to understand the changes in the parts below – otherwise, let me know if you're in doubt about any of it.. 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 

Thanks for all the time you put into the wiki – it's really an amazing place to search for knowledge.. !! :)

 

Have a Greay day ..! :)

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

SeagullUAV, 2015-11-03:
Hi Hamish,

 

I don't know if it's too much to ask, but could Seagull #MAP be placed on top here since we are the newest product to market.. ?

 

http://planner.ardupilot.com/wiki/common-camera-shutter-triggering-cables-for-purchase/

 

I know for a fact that Smart Little Things that produces the Stratosnapper did get declared bankrupt only few days ago here in Denmark.

Really sad, since they really made some awesome stuff..! Let's hope they make it !!

Link to the sad news: http://www.dr.dk/nyheder/regionale/bornholm/little-smart-things-droner-er-landet-i-en-konkurs

(sorry – it's in Danish)

 

Have a nice day !

 

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

hamishwillee, 2015-11-03:
Hi Rone

Sad news for Stratosnapper :-) I have reordered with MAP first - purely on the basis of alphabetic order for my naming convention. This is not set in stone - if the wiki owners come along with a preferred order then I would switch again.

If you see above, your images did not come through, so I cannot make any changes. You're probably going to have to go onto github directly if you're adding screenshots.

Regards
Hamish

SeagullUAV, 2015-11-03:
Hi Hamish,

 

Sorry about the missing images..! :/

 

You can access all the changes incl. the 2 new pictures here:

 

http://www.seagulluav.com/hamish/

 

To view the page, enter the password:  ardupilot

 

Hope this will work then.. :)

 

 

And thank you for the change in order..! :) It means a lot for a new startup like Seagull UAV – we're only just about 3 months old now..! :)

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

hamishwillee, 2015-11-03:
I made the updates. The only problem is the configuration - the supplied mission planner picture and the shutter duration differ - it is not clear which one is accurate.

SeagullUAV, 2015-11-03:
Hi Hamish,

 

Ohh sorry..! It's getting late here.. ;)

 

Your text was OK – my bad.! Check the link again – added the updated photo with a clear UPDATED text above it.. :)

 

Best regards

Rone Scholz-Nielsen
CEO // founder

Proudly based in Denmark

Visit the Shop at  <https://seagulluav.com/> SeagullUAV.com

Like and follow: 

 <https://facebook.com/seagulluav>   <https://twitter.com/seagulluav>   <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

hamishwillee, 2015-11-04:
I think we're all done then! Closing this issue :-)

SeagullUAV, 2016-04-29:
Hi Hamish,

We no longer sell Seagull #MAP, since it have been replaced by Seagull #MAP2 with even better features for our users..

If you feel for it, we would love to have the wiki updated with the new version..

I'm looking forward to your reply..

Have a Great day ! ☺

Best regards
Rone Scholz-Nielsen
CEO // founder
[cid:CC0364F0-E515-4B33-9810-A81B4075CADD]
Proudly based in Denmark
Visit the Shop at www.SeagullUAV.com<http://www.seagulluav.com/>

Like and follow:
[facebook_icon_24x24_bw]<https://facebook.com/seagulluav> [twitter_icon_24x24_bw] <https://twitter.com/seagulluav>  [linkedin_icon_24x24_bw] <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

Closed #197<https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/197>.

magicrub, 2016-04-29:
What would you like it to say?

SeagullUAV, 2016-04-30:
Hi Tom,

The site more or less needs to be updated with info on Seagull #MAP2 instead…

Hamish did the initial setup for our first #MAP.. It just needs to be updated with the new features, new links and images, and off course the updated support..

Let me know if you need any further information..

Best regards
Rone Scholz-Nielsen
CEO // founder
[cid:CC0364F0-E515-4B33-9810-A81B4075CADD]
Proudly based in Denmark
Visit the Shop at www.SeagullUAV.com<http://www.seagulluav.com/>

Like and follow:
[facebook_icon_24x24_bw]<https://facebook.com/seagulluav> [twitter_icon_24x24_bw] <https://twitter.com/seagulluav>  [linkedin_icon_24x24_bw] <https://www.linkedin.com/company/10157090?trk=tyah&trkInfo=clickedVertical%3Acompany%2CclickedEntityId%3A10157090%2Cidx%3A1-1-1%2CtarId%3A1443695309118%2Ctas%3Aseagull%20uav>

hamishwillee, 2016-04-30:
@Thanks! Best to create a new issue for new work please @SeagullUAV - if you reopen old pages keeping track of what has been done/and has not been done is hard.

Given the 2 products are very similar I propose that you update the article to refer to #MAP2 - perhaps adding a note that this supercedes #MAP1 which is no longer in production. Then submit this as a PR for our review. The changes should be fairly minor. Would that be OK?

closed 2016-04-30 by hamishwillee