ArduPilot/ardupilot_wiki #200: Motor Test description link

Fossdal, 2015-11-02:
In MissionPlanner under "Initial Setup"/ "Motor Test" there is an help tekst 
"Please click here to see your motor numbers, scroll to the bottom of the page"

if i click on the link i come to an "Not Found Page"
the link is http://copter.ardupilot.com/wiki/motor-setup/

hamishwillee, 2015-11-03:
@Fossdal I cannot find the URL of the page that contains that bad link. Can you please provide the actual URL?

Fossdal, 2015-11-05:
Sorry this is not an error on the Wiki, it is an error in MissionPlanner software pointing to wrong url
it is in 
https://github.com/diydrones/MissionPlanner/blob/master/GCSViews/ConfigurationView/ConfigMotorTest.cs#L168

closed 2015-11-05 by Fossdal

CraigElder, 2015-11-07:
@meee1