ArduPilot/ardupilot_wiki #202: Add clones to wiki

rmackay9, 2015-11-03:


hamishwillee, 2015-11-03:
Happy to do so. I think we discussed this previously and decided that we'd only have dronecode members, and by implication, only those with published schematics etc.

@CraigElder @rmackay9 Did you have a "seed list" of such manufacturers/boards? I can probably pull a lot of the other information together once I have a set of names/urls for purchasing the products. If not I can search and present initial list to you.

CraigElder, 2015-11-03:
We have the list of boards that support the project and the boards supported by the project here. http://dev.ardupilot.com/wiki/supported-autopilot-controller-boards/.

hamishwillee, 2015-11-03:
That is true - after all, I created it :-) @rmackay9 Are there other "clones"/information you were expecting?
Not all of these have schematics (or at least I couldn't find them last time I looked)

CraigElder, 2015-11-03:
The supported ones have schematics and are open source

hamishwillee, 2015-11-03:
@rmackay9 Closing this then - please reopen if you see a board missing or know if any other issues/known problems that could be added.

closed 2015-11-03 by hamishwillee