ArduPilot/ardupilot_wiki #203: Simple/Super Simple Mode on Copter

lvale, 2015-11-05:
Hi

I was reviewing the code at 
https://github.com/diydrones/ardupilot/blob/master/ArduCopter/switches.cpp#L275

and the code does the following when a CH7/CH12 has SuperSimple Mode enabled:

On a 2 position switch it can be either off or SuperSimple mode enabled.

On a 3 position switch it can be off, middle switch is SIMPLE, high is SUPERSIMPLE.

The following page should be more explicit regarding this http://copter.ardupilot.com/wiki/channel-7-and-8-options/

And this paragraph :
Ch7 or Ch8 can be set-up to enable/disable Simple and Super Simple mode.
If Simple Mode is selected the raising the switch high will enable Simple mode, low will disable it.
If Super Simple mode is selected then raising the switch high will enable Super Simple mode, low will disable it and if a 3-position switch is used the middle position will enable regular simple mode.
Only Ch7 or Ch8 (not both) should be set to either Simple mode or Super Simple mode.
The Ch7/Ch8 switch position will be override the simple/super-simple options selected on the Flight Modes screen.

here:http://copter.ardupilot.com/wiki/simpleandsuper-simple-modes/#use_ch7ch8_to_select_normal_simple_or_super_simple_mode

could be a bit more clear :)

btw: can we rename the CH7/CH8 channels to something else like: OPTIONS CHANNELS ? because now we can go up to CH12 ?

hamishwillee, 2015-11-06:
@lvale Thanks very much for the comments - good points. @rmackay9 - I need your  input/review on this.

1. Renaming. I think this makes a lot of sense. Perhaps "Auxiliary Channel/Switch Options"?
1.  [Channel 7 and 8 Options](http://copter.ardupilot.com/wiki/channel-7-and-8-options/) ([diff](http://copter.ardupilot.com/wp-admin/revision.php?from=16258&to=16261)) - please check
1. [Simple and Super Simple Modes](http://copter.ardupilot.com/wiki/simpleandsuper-simple-modes/#use_ch7ch8_to_select_normal_simple_or_super_simple_mode) ([diff](http://copter.ardupilot.com/wp-admin/revision.php?from=15690&to=16262)) - please check

So basically, is this OK, and can we agree on above name or another name?

rmackay9, 2015-11-06:
I prefer "Auxiliary" to appear in the name because that's what we call them in the code.  So Auxiliary Switches, Auxiliary Function Switches, etc.
The changes look good to me.

hamishwillee, 2015-11-06:
"Auxiliary Function Switches" it now is! Thanks @rmackay9 @lvale . Closing

lvale, 2015-11-06:
@hamishwillee @rmackay9 

On this page I believe that the Landing Gear entry on the table should be revised, because if Landing Gear is on a 3pos switch Auto will work for LAND and RTL when in the middle position. From the landinggear.cpp

       if (control_mode == LAND ||
           (control_mode==RTL && (rtl_state == RTL_LoiterAtHome || rtl_state == RTL_Land || rtl_state == RTL_FinalDescent)) ||
           (control_mode == AUTO && auto_mode == Auto_Land) ||
           (control_mode == AUTO && auto_mode == Auto_RTL && (rtl_state == RTL_LoiterAtHome || rtl_state == RTL_Land || rtl_state == RTL_FinalDescent))) {
            landinggear.force_deploy(true);
        }

from the Landing Gear page
Set Ch7 or Ch8 Opt to “Landing Gear” in order to enable manual control.  When the switch is in the “low” position the gear will be deployed, “high” will be retracted, “middle” will be a neutral position that will not change state (will be used for automatic operation on a future release).

hamishwillee, 2015-11-10:
@lvale I got this message via email, but it doesn't appear here (very odd):

> @hamishwillee @rmackay9

>  On this page I believe that the Landing Gear entry on the table should be revised, because if Landing Gear is on a 3pos switch Auto will work for LAND and RTL when in the middle position. From the landinggear.cpp
> 
   ```
   if (control_mode == LAND ||
       (control_mode==RTL && (rtl_state == RTL_LoiterAtHome || rtl_state == RTL_Land || rtl_state == RTL_FinalDescent)) ||
       (control_mode == AUTO && auto_mode == Auto_Land) ||
       (control_mode == AUTO && auto_mode == Auto_RTL && (rtl_state == RTL_LoiterAtHome || rtl_state == RTL_Land || rtl_state == RTL_FinalDescent))) {
        landinggear.force_deploy(true);
    }
```
> from the Landing Gear page
Set Ch7 or Ch8 Opt to “Landing Gear” in order to enable manual control. When the switch is in the “low” position the gear will be deployed, “high” will be retracted, “middle” will be a neutral position that will not change state (will be used for automatic operation on a future release).

Can you please confirm that you want to update the [Auxiliary Function Switches](http://copter.ardupilot.com/wiki/channel-7-and-8-options/) section on landing gear, which currently says?:

> Landing Gear:	Retracts/Deploys landing gear. (AC3.3 only)

Would you be happy with the following  text?:

> Note: This feature was introduced in AC3.3

> This channel deploys/retracts the landing gear based in response to the switch low/high position. If mapped to a three-position switch the middle position is used to specify that the landing gear should auto-deploy and retract in AUTO mode.

In addition, if the auto feature is now implemented we should update the [Landing Gear](http://copter.ardupilot.com/wiki/landing-gear/) text you point to above to remove info on "future release". @rmackay9 IN what version did the support for "AUTO" landing mode on the middle switch appear (please confirm that it is actually present).

P.S. Luis, really good to open a new issue if you have a different problem after the original issue is closed.

lvale, 2015-11-10:
sorry @hamishwillee I did post after closed but then retracted.... :)

hamishwillee, 2015-11-10:
@lvale So this change isn't valid/needed?

lvale, 2015-11-11:
Yes it is valid and needed, and your suggestions are ok. I only retracted because I had an extra commit for the landing gear but it still needs some rework.

hamishwillee, 2015-11-11:
OK, so @rmackay9 when did the auto-landing-gear stuff go in (docs say manual stuff in 3.3, but identify auto stuff as "future release)

closed 2016-03-28 by lvale