ArduPilot/ardupilot_wiki #207: Making a Mavlink WiFi bridge using the Raspberry Pi - using usb port instead of uart

tommy-b-10, 2015-11-08:
[Making a Mavlink WiFi bridge using the Raspberry Pi](http://dev.ardupilot.com/wiki/making-a-mavlink-wifi-bridge-using-the-raspberry-pi/) - 

You can now use the usb port of the radio and raspberry pi instead of the serial uart connections (making it much easier to get setup) Just change all instances of ttyAMA0 to ttyUSB0 :D

hamishwillee, 2015-11-09:
Hi @tommy-b-10 

I'm hesitant to update the docs without being able to try this. Can you expand a little on the setup?

1. I think that what you're suggesting is in section [Connecting a 3DR Telemetry Radio to a Raspberry Pi](http://dev.ardupilot.com/wiki/making-a-mavlink-wifi-bridge-using-the-raspberry-pi/#connecting_a_3dr_telemetry_radio_to_a_raspberry_pi) we use a USB connection rather than that complicated wiring to the radio

  * Is that correct?
  * I see that the 3DR Radio does indeed have a usb connection ... when did this appear?

2. Then in section "Configuring MavProxy to always run and listen to incoming connections" I think you're suggesting changing the mavgateway link from
```
DAEMON_ARGS="--master=/dev/ttyAMA0,57600 --out=udpin:0.0.0.0:14550 --daemon"
```
to
```
DAEMON_ARGS="--master=/dev/ttyUSB0,57600 --out=udpin:0.0.0.0:14550 --daemon"
```
* Is that really all you need to do to get connection to radio over USB working?
* Is the baud rate still OK?
* Are there any "costs" to this move. 
* Any idea where the mavgateway script went to (I can find if needed, just hoping you can save some time).

Thanks for your help!

tommy-b-10, 2015-11-09:
Hi @hamishwillee 

No problem, I'll do my best!

1. Yes that is correct, I'm not sure when the radio got USB, but I am using a clone version from ebay. This should still run the same.

2.
* That is all you need yes
* Baud rate is fine, but depends on what your radio baud rate is. Mine is 115200 as my APM board wont accept 57600 for some reason.
* Shouldnt be any costs, maybe the tiniest bit of latency, but I cant see why it would be any issue.
* They removed the mavgateway script from the main dronekit API, but I managed to find it by googling it. Here is [LINK](https://github.com/dronekit/dronekit-python/blob/3222b2cdb71b8b4c5b924ee28fc5d3529be00e67/scripts/mavgateway)
, I suggest rehosting it here so you can make sure it is kept.

P.S. The raspberry pi image you have linked at the end is actually too large to fit on the 8gb SD card I have (its like 500kb too large :P), I'd recommend not expanding the file system if you are going to be giving it out, so it dosent give anyone else any issues using an 8gb card.
Any other questions, please ask!

hamishwillee, 2015-11-10:
Hi @tommy-b-10 

Thanks very much. I'll look into updating this in the coming week. Hope you'll be able to review the changes!

Re the 8Gb card - would best thing be to suggest that someone get a 16GB card?

Cheers
H

tommy-b-10, 2015-11-10:
@hamishwillee 

No problem, I'd be happy too!

Most people only use 8gb cards for raspberry pi's cause they are cheaper. If you are redoing the instructions, perhaps you could upload a new image? Just skip the expand filesystem step :) If not, people with 8gb cards will just have to follow the instructions :P

Thanks!

hamishwillee, 2015-11-11:
Hi @tommy-b-10 

Thanks. I'm not doing a proper update because my RPi died and has not yet been replaced :-( What I have done is some "spot" updates:

* Fixed the broken mavegageway file link
* Tidied up the installation of MAVProxy section - this used old droneapi instructions and was very out of date.
* for the [sd_card_image_file](http://dev.ardupilot.com/wiki/making-a-mavlink-wifi-bridge-using-the-raspberry-pi/#sd_card_image_file) section I changed the text to 

  > To restore **this image** to an SD Card you will need an SD Card with at least 16GB of space, ...

  Obviously it would be better to build/upload a new image, but as above, not going that far. I'll accept a new one from someone else!
* I've added this note to make it clear that USB is an option: 
[image]
* I've added this bit to make it clear how to configure for USB:
[image]

This now captures the information, albeit not in a perfect form. I'm now "parking" this. My next step will be to update the "tip" above with a short separate section about setting up USB and an image.

Thanks for all your help.

tommy-b-10, 2015-11-11:
@hamishwillee 

Looks good! However, with the baud rate, it is entirely dependent on the baud rate you have set using the software. By default 57600 is set, however you can change it in the mission planner software. So perhaps you should keep the baud rate in the instructions at 57600 for usb aswell.

Cheers!

hamishwillee, 2015-11-13:
@tommy-b-10 I've updated it as you suggested. I thought the rate was dependent on the connection type. By "software" do you mean config software for the telemetry radio itself, and if not, what?

davidbuzz, 2016-11-15:
this has made it into the docs as a "TIP" for using USB-serial as an alternate to hardware serial.

closed 2016-11-15 by davidbuzz