ArduPilot/ardupilot_wiki #208: Misinformation on common-Maxbotic Analog Sonar – Copter3.2

tschundler, 2015-11-08:
[common-Maxbotic Analog Sonar – Copter3.2](http://copter.ardupilot.com/wiki/common-rangefinder-maxbotix-analog/) 

This page says to set the meters per volt value to 2.04 because that is 7m (range) / 5v (power supply), only 7/5 is 1.4, not 2.04, so this value looks incorrect.

However, the 2.04 number is correct, because it is derived from the Vcc/1024 volts per cm value in the datasheet for the Matbotix sonar modules. If Vcc = 5, 5/1024 = 0.00488v/cm = 0.488v/m = 2.04m/v

See also #81 - the sonar pages as a whole are confusing and need to be cross-referenced better to understand why there are three separate ones.

hamishwillee, 2015-11-10:
Hi @tschundler 

Thanks for this. I've come into a busy two week period, but I have assigned myself this and will look at it as soon as I can.

gmorph, 2016-11-15:
The sonar page has been updated with a Note describing the math calculation above to get the 2.04 number.  Thanks.

closed 2016-11-15 by gmorph