ArduPilot/ardupilot_wiki #210: Add motor thrust page to copter side menu

rmackay9, 2015-11-09:
I've created this new motor thrust wiki page.
http://copter.ardupilot.com/wiki/motor-thrust-scaling/

Could you add it to the side menu?

hamishwillee, 2015-11-09:
Hi Randy,

I put it under "Tuning" (in Advanced Configuration) - is that OK?

The two parameters MOT_THST_EXPO and MOT_THST_MAX are not in the parameter list (I like to link parameters if possible). Is there a reason for this?

Other thoughts. Feel free to ignore ...

* The image would be better if it went with MOT_THST_EXPO from low to high, rather than mid-low-high. 
I*s the linear/stright line curve labled "beforex" in each case just a representation of a perfectly linear ESC thrust output?
*  Would probably make sense for the index labels not to have the numbers and for that to be clear. What would be good here is a diagram showing a real thrust curve and how closely the default parameters match it - e.g. for Iris say.
* Also, any advice on tuning?

rmackay9, 2015-11-09:
Hamish,
Yup, under tuning is fine with me.  I've reordered the pictures and moved them down to the mot-thst-expo parameter description.
The "before" is really the un-adjusted output to the motor.
Leonard's got a bit of mavlink that I'll add to the bottom of the page that might help people figure out what the parameters should be if they've gone to the trouble to measure the vehicle's actual thrust curve.  It's a bit unlikely that many individual people will do this but RTF manufacturers and a few really dedicated individuals might.

hamishwillee, 2015-11-10:
Hi Randy

Thanks for the changes - certainly helpful. I've made the additional note you added into a "tip" near the top to highlight that this isn't something most people will need to do. I also changed the code markup into a single "pre".

Should the MOT_THST_EXPO parameters be visible in the parameters list?

Cheers
H

rmackay9, 2015-11-10:
@hamishwillee, yes, the MOT_THST_EXPO param description should be visible.  I think it's not because our parameter description parsing stuff is confused by the appearance of other frame types (like helicopter).  I think we will need to remove the param description info for heli for the moment.

hamishwillee, 2015-11-10:
@rmackay9 Thanks. Can you let me know when that's done, and I'll link.

rmackay9, 2018-12-26:
this is done so closing

closed 2018-12-26 by rmackay9