ArduPilot/ardupilot_wiki #214: Submitting Patches Back to Master

lvale, 2015-11-10:
I believe the following text could be added to this section:

After a recent effort of revising all the messages sent by ArduPilot to the Ground Stations, and defining some uniformity of their severity according to MAV_SEVERITY (https://pixhawk.ethz.ch/mavlink/) where this indicates the severity level, generally used for status messages to indicate their relative urgency. Based on RFC-5424 using expanded definitions at: http://www.kiwisyslog.com/kb/info:-syslog-message-levels/, it would be useful that, future developments on the code base could benefit from the current work, and use the appropriate severity for the message being sent.

There are three main mechanisms of sending messages: gcs_send_text() and gcs_send_text_fmt(), and a third that broadcasts to all active Mavlink connections: send_statustext_all().

The main difference between the first two is that gcs_send_text_fmt() sends a low priority formatted message to the GCS and only one fits in the queue, so if you send more than one before the last one gets into the serial buffer then the old one will be lost. The third mechanism sends to all listening stations a message with a specified severity.

davidbuzz, 2016-11-15:
I'm not sure I see how this will work.  closing.   please re-open if you think this hold value.

closed 2016-11-15 by davidbuzz

lvale, 2016-11-17:
@davidbuzz Thanks. It's so old I forgot it, and lost it's purpose....