ArduPilot/ardupilot_wiki #215: add 4Gmetry to companion computer wiki page

rmackay9, 2015-11-13:
We should create a little list of known working companion computers.

This one for example is setup to work with ardupilot straight out of the box (I think)
     http://4gmetry.voltarobots.com/

rmackay9, 2015-11-13:
By the way, on the Edison for Drones page we should have a link to sparkfun to show where to buy it and which additional carrier boards are required.  I think it's these 3 but Jaime can confirm.
https://www.sparkfun.com/products/13024
https://www.sparkfun.com/products/13040
https://www.sparkfun.com/products/13045

hamishwillee, 2015-11-13:
@rmackay9 Agreed. How do you find out about these/how do we populate the list. This will look pretty silly if I create a page with just one link.

I've updated http://dev.ardupilot.com/wiki/edison-for-drones/ with new title and slightly improved structure. Which Jaime should I chase for this information?

rmackay9, 2015-11-13:
Thanks for that.  I'm not sure how we will build up the list.  My guess is that people will slowly tell us about companion computers as they want to advertise them.  I guess the landing page is automatically generated so we can't add it there right?

it's Jaime Machuca, I've ping'd him a note on skype so hopefully he'll see tis.

jmachuca77, 2015-11-13:
yeah, those three modules are the basic ones you need, you may also need to add:

https://www.sparkfun.com/products/13041

In case you want more storage, I found once you start installing stuff you run out pretty quickly

hamishwillee, 2015-11-15:
Hi @rmackay9 (thanks @jmachuca77 !)

I've added purchasing information for Edison [in the overview here](http://dev.ardupilot.com/wiki/edison-for-drones/#overview). Please check and let me know if you think this needs to be separated into its own section.

With respect to the companion computer list, what I have done is created [Turnkey Companion Computer Solutions](http://dev.ardupilot.com/wiki/turnkey-companion-computer-solutions/) which people can extend as they wish. Yes it is a bit empty, but we have to start somewhere :-) Can you please review, and if it is OK share with the wider community (I will do this too, but I figure since you're already having a discussion about companion computers it makes sense that you push this now.

rmackay9, 2015-11-16:
@hamishwillee, looks good.  I slightly modified the the wording to remove "perfect".  Maybe I shouldn't worry and people will expect that the description comes from the manufacturer... but I removed it anyway.

hamishwillee, 2015-11-16:
@rmackay9 Good catch - we shouldn't put in stuff that is clearly too "marketing". Closing.

closed 2015-11-16 by hamishwillee