ArduPilot/ardupilot_wiki #219: PX4FMU Wiring Only - Page Disappeared

billbonney, 2015-11-18:
I cannot find the page http://copter.ardupilot.com/wiki/px4fmu-only-wiring/

which is related to this diagram http://copter.ardupilot.com/wp-content/uploads/sites/2/2013/04/PX4FMUwire1.jpg

Can it be put back, or put in an Appendix, thx.

billbonney, 2015-11-18:
The image and instructions AFAR as I cannot see the old page any more is completely different. One is for creating a quadcopter using the PX4FMU only, the other is using the PX4FMUv1 with this PX4IO daughter board. I don't understand why is was deleted (though it seems it was misinterpreted and being a duplicate)

hamishwillee, 2015-11-18:
@billbonney The delete approval from @rmackay9 was:

>   The page has a lot of duplication with other pages so I’m fine deleting it.
 
>  The thing that’s unique about this page is that it tries to describe how a px4 fmu 
    (i.e. the top only) can be used to fly a quad.  It’s slightly interesting because 
    it allows a very small flight controller but in fact, it’s not a method ardupilot 
    supports anymore and it looks like 3DR doesn’t sell the “rainbow cable” that’s 
    required to actually implement this.  This means I think nobody’s actually doing this.
 
>   If people want a small flight controller – arsovtech (a dronecode member) has created a small pixhawk based boards that is much more popular with the community.              http://arsovtech.com/?p=1447

@rmackay9  Thoughts? I can probably recover if necessary from an archive.

billbonney, 2015-11-18:
I just had a support request asking how to do this. I think it should have been archived (not deleted) since the changes to support it again are not huge. I know there are other small boards available. But we should be careful about throwing away learnt knowledge just because we have moved on.

rmackay9, 2015-11-19:
At some point we need to delete stuff that we don't think it useful anymore or that is more difficult to maintain than it's worth considering the number of people who will actually try to do it.  It's been gone for 5 months and nobody's noticed.. it's likely the instructions would be out of date even if they did still exist.  ..6 weeks from now it's not even going to be possible to buy these boards.

I think asking on drones-discuss is probable the way to go.  I believe it's a config change and/or the upcoming changes to support the px4racer may also work.

billbonney, 2015-11-19:
Why do we need to delete the page and not keep it in an Archive Section (or just extract the relevant info and keep in an archive section). I still have PX4v1FMU boards and so do many others.. so it's still useful. We should be careful about what is thrown away. The schematics are still Open Source HW so still relevant for somebody designing a new new FC. It's an easy way to reduce the cost of the BOM.

rmackay9, 2015-11-19:
Sure, if you feel strongly about it and hamish wants to bring it back and stick it in the archive section that's fine with me.  That way, although it's likely it will never be required again, it's still there to be found.

hamishwillee, 2015-11-19:
I'll see what I can do. 

We can of course archive everything, but there is a tendency for it to clutter up search and start throwing up bad link errors all the time. If something is really useless it is better to remove it. Drawing the line is tough.

hamishwillee, 2015-11-19:
I've re-added the page at the old url (source from my [sneeky github archive](https://github.com/diydrones/ardupilot_wordpress_sources/blob/820ce9e504956921a798399508dfded8d141e24c/copter.ardupilot.com/px4fmu-only-wiring.html)) and archived. Broken links have just been removed (text LINK_BROKEN) in place.

Closed!

closed 2015-11-19 by hamishwillee