ArduPilot/ardupilot_wiki #224: Building for Bebop on Linux: Recovery Tip

djnugent, 2015-11-21:
[Building for Bebop on Linux](http://dev.ardupilot.com/wiki/building-for-bebop-on-linux/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

Tip for **Recovery**:
If you are getting a read-only file system error while trying to restore your rcS init file, then try running `mount -o remount,rw 'rootfs'` to make your file system writeable.

@jberaud is this reasonable?

jberaud, 2015-11-21:
mount -o remount,rw /​
It's totally reasonable. I'll do that along with the toolchain issue.

Julien

hamishwillee, 2015-11-22:
@jberaud I saw you made this change, and pressed publish.

The change is shown below:

[image]

This is a bit different than what @djnugent was suggesting - ie his was like a tip to address the specific case of recovery and specifies 'rootfs". Can you confirm it meets the requirement?

jberaud, 2015-11-23:
@hamishwillee  It's exactly equivalent on the bebop. Thanks

hamishwillee, 2015-11-23:
Great, then I am closing this.

closed 2015-11-23 by hamishwillee