ArduPilot/ardupilot_wiki #225: common-LightWare SF10 Lidar

DrFoxxer, 2015-11-23:
[common-LightWare SF10 Lidar](http://copter.ardupilot.com/wiki/common-lightware-sf10-lidar/) 

After following all available instructions I cannot get the SF10 C to display altitude.
I have chosen the serial connection to start with and haven't tried I2C or analog.
The lightware terminal software works well but Pixhawk and the SF10C don't want to talk.

hamishwillee, 2015-11-23:
@DrFoxxer Thanks for letting us know. Can you please raise on the APM forums and cross link to here. The forums are more intended for technical support - and if the problem proves to be with the documentation we will then update it.

DrFoxxer, 2015-11-24:
@hamishwillee Thank you. I will post on APM forums. After working with ardupilot for almost 3 years this is the first time that I am reaching out for help so my apologies for posting here.

hamishwillee, 2015-11-24:
@DrFoxxer Not a problem. That is a very new article, and I am hoping that it is something to do with your hardware and not the docs!

CraigElder, 2015-11-24:
@DrFoxxer This is some sort of user error.  I have installed a dozen of these units and use them almost daily. I don't see any issues with the instructions.

hamishwillee, 2015-11-24:
@DrFoxxer I'll close this then - if Craig has validated the instructions that is good enough for me. Hope the forums are able to help you.

closed 2015-11-24 by hamishwillee