ArduPilot/ardupilot_wiki #226: Mission Planner needs docs on network connections

djnugent, 2015-11-23:
[common-Connect Mission Planner to AutoPilot](http://planner.ardupilot.com/wiki/common-connect-mission-planner-autopilot/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

With mavproxy and network connected drones becoming more common, we should update MP docs on how to connect over UDP, TCP, UDPCI. 

This stems from an issue(https://github.com/diydrones/MissionPlanner/issues/1064) I had where my firewall was blocking MP from receiving UDP connections.

hamishwillee, 2015-11-23:
@djnugent I'm not sure I agree with this. The connections are set up in pretty much the same way in each case so to my mind covering each connection case in some level of detail would be repetitive and not particularly useful.

I think the right way to deal with this is to add a specific troubleshooting issue for the firewall case. If we get a lot of issues for a particular type of connection, then that would be a trigger for creating a separate section.

I've added this line in [troubleshooting](http://planner.ardupilot.com/wiki/common-connect-mission-planner-autopilot/#troubleshooting):

[image]

I'd like to have added information on how you unblock the firewall, but I think that would be very platform user configuration dependent. Better to highlight the issue and let them use google to sort it out for their configuration.

I think this works, but if you have advice on how we might better restructure let me know.

OK?

djnugent, 2015-11-23:
That works for me. Thanks.

closed 2015-11-23 by hamishwillee