ArduPilot/ardupilot_wiki #229: "make sitl -j4" command returns fatal error!

pil0706, 2015-11-24:
i followed all the directions "http://dev.ardupilot.com/wiki/sitl-native-on-windows/" here and i added "export PATH=$PATH:$HOME/ardupilot/Tools/autotest" to the end of .bashrc

[image]

and i typed "make sitl -j4" and it returns...

[image]

there are some foreign languages but if you are the well trained you would be okay to understand what this tells to you.. (i have no idea what this means even in my mother language)

if you guys have had this kind of issue and solved, please let me know how to fix this..

hamishwillee, 2015-11-24:
@pil0706 This channel is for issues with the documentation. I have verified this article personally against Windows 8.1 and Windows 10 (clean installation) so I know that it is the correct setup.

As I can't reproduce the problem (see #195 for another instance of the same problem) there is not much I can do to help. One suggestion is that you might have more luck building with Vagrant!

Please raise this request on the APM Forums, where more people with build expertise may be able to help you.

closed 2015-11-24 by hamishwillee

pil0706, 2015-11-25:
@hamishwillee 
Hey, i've checked the #195 already. but maybe i was lost in somewhere. I will following #195 again see if it works

thanks for reply!

hamishwillee, 2015-11-25:
@pil0706 #195 did not resolve the issue. It was closed because the documents worked in my test case. The right "next step" is to seek advice on the APM forums.

pil0706, 2015-11-25:
Hey, it works now!

hamishwillee, 2015-11-25:
Great. Closed.