ArduPilot/ardupilot_wiki #230: common-Measuring Vibration - out of date warning can be removed

peterbarker, 2015-11-25:
[common-Measuring Vibration](http://copter.ardupilot.com/wiki/common-measuring-vibration/) - there is a disclaimer on this page indicating the code is not yet live in the stable distribution which is out-of-date.

hamishwillee, 2015-11-25:
@peterbarker So are the changes present in copter, rover, plane current stable releases?

hamishwillee, 2015-11-25:
@gmorph There is a note against [common-Measuring Vibration](http://copter.ardupilot.com/wiki/common-measuring-vibration/) that says:

> At time of writing (27 July 2015), Copter-3.3 is still a release candidate. Its changes are not yet present the official release of Copter, Plane and Rover. The changes here are expected to be in the next release of Plane and Rover.

Did the changes go into Plane/Rover? If so, what version?

gmorph, 2015-11-26:
The changes are in Plane 3.4.
I haven't done a new release of Rover yet so its still not available.

Thanks, Grant.

hamishwillee, 2015-11-26:
Thanks @gmorph. I've updated note as shown:

[image]

Closing - now done.

closed 2015-11-26 by hamishwillee