ArduPilot/ardupilot_wiki #232: update Copter GPS failsafe docs

rmackay9, 2015-11-28:
The docs talk about a separate GPS failsafe but this has been merged into the EKF failsafe so we should update the docs.
http://copter.ardupilot.com/wiki/gps-and-gcs-failsafes/

hamishwillee, 2015-11-28:
@rmackay9 Happy to do this, but "how" has this been merged and in what version?

If I had to guess I'd say the merge is more like "replace" - ie can I assume that FS_GPS_ENABLE  is no longer used at all and the behaviour is as defined in [EKF / DCM Check & Failsafe](http://copter.ardupilot.com/wiki/ekf-inav-failsafe/)?

If so, a couple of extra questions:

1. The EKF docs say "The EKF check runs only on the Pixhawk " So I presume you're actually saying "it doesn't run on APM"? Presumably it runs on a board that it fits on?
1. Also "and only when the EKF is being used as the primary source for attitude and position estimates (i.e. AHRS_EKF_USE = 1). " So are you saying now that AHRS_EKF_USE  replaces FS_GPS_ENABLE ? Presumably EKF is always "used as the primary source ..." now?

Presumably there will still be people using the GPS failsafe? Would the right thing be to split that out into its own document and archive?

Thanks!

hamishwillee, 2015-11-30:
@rmackay9 I started this work, but need you to check it and confirm.

1. Created/split off and archived [Archived: GPS Failsafe](http://copter.ardupilot.com/wiki/archived-gps-failsafe/).

   * This states support for GPS Failsafe merged in Copter 3.3. Please confirm version.
   * ISSUE - this links to [GPS Failsafe and Glitch Protection](http://copter.ardupilot.com/wiki/gps-failsafe-glitch-protection/). Is the article still relevant, and if so, how should it be changed? I'm guessing that the [EKF section](http://copter.ardupilot.com/wiki/gps-failsafe-glitch-protection/#glitch_protection_8211_ekf) is now "standard" - perhaps we should push the GPS specific information into the GPS Failsafe doc, and rename this document "EKF Failsafe and Glitch Protection?"

2. Renamed the original topic [GCS Failsafe](http://copter.ardupilot.com/wiki/gcs-failsafe/) and updated.

3. Added note to the [EKF / DCM Check & Failsafe](http://copter.ardupilot.com/wiki/ekf-inav-failsafe/) "Starting in Copter 3.3 the EKF failsafe replaces the <a href="archived-gps-failsafe">GPS Failsafe</a>."

4. Fixed up all links to the (now split) document.

5. In [Simple Geofence](http://copter.ardupilot.com/wiki/ac2_simple_geofence/) we have statements like "If GPS failsafe is enabled and the Fence ...". Can these be replaced directly by "EKF failsafe"?

6. The AHRS_EKF_USE and EKF_CHECK_THRESH parameters are not showing up in the [Copter Parameters](http://copter.ardupilot.com/wiki/arducopter-parameters/). I have created an issue for this: https://github.com/diydrones/ardupilot/issues/3248